prusa3d/PrusaSlicer #1880: [Feature request] Move octoprint settings out of "printer"

Spacemarine2018, 2019-03-01:
### Version
1.42.0-alpha 5

The settings for the connection to octoprint are located in the "printer" tab. This is not ideal for several cases:

- For every nozzle size I use, I have to save a "new" printer with the according octoprint settings, since the original profiles can't be overwritten.

- If I import a printer setting from someone else, I need to modify his profile and save it to make it compatible with octoprint

- If I import a large model and make a lot of adjustments and later decide to send it via octoprint instead of SD-card profile, all my previous changes will be gone once I switch the printer to a octoprint printer.

Instead, I think it should be solved in the following way:

- Move the the octoprint setting out of the printer setting into something like the general configuration - preferences
- Make multiple different connections possible to different octoprint hosts
- Give the user the option to select different octoprint hosts under "send to printer"

supermerill, 2019-03-01:
Makes sense. I don't see why i need to duplicate all my printer settings because i have two (identical) mk2.

bubnikv, 2019-03-01:
We plan to introduce a concept of printer device. You will have the option to define the connection settings for the printer device, and to link a printer profile to the printer device.

We will not manage to implement this concept in this release though.

Spacemarine2018, 2019-03-01:
I hope that one printer profile can be linked to multiple printer devices?
This would be useful for the people with many identical printers.

bubnikv, 2019-03-01:
that is the idea.

reinhapa, 2019-03-01:
I would be perfectly happy to choose a printer at the time I press the „send to printer“

bubnikv, 2020-10-17:
Physical printer profiles are implemtned in PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-17 by bubnikv