prusa3d/PrusaSlicer #1882: [UX Improvement] Plug-and-play by using USB and Zeroconf in the configuration wizard

probonopd, 2019-03-01:
With more and more Prusa printer variations out there, the set up wizard grows larger and larger in each version. Hence I propose the following to make for a better "plug-and-play" experience:

* The printer firmware should be able to describe the exact printer model
* The exact printer model should be send in Zeroconf advertisements as a TXT message (possibly the OctoPrint protocol should be extended for this)
* When the initial configuration wizard wizard comes up, it scans for all USB devices and Zeroconf without user interaction, and shows any found printers automatically. The user does not need to select a model
* Only if no printers are found over USB and Zeroconf, present the user with a printer selection window. (Maybe the printer is not connected and only used with SD card)

This scheme could even be extended with the printer describing its properties, e.g,. bed size, supported speeds etc. rather than just advertising the model.

I think that this workflow would be much closer to the vision of Zeroconf than what we have now, with what was supposed to be a "plug and play" feature being hidden away under "Advanced" in the last tab under "Printer Settings". 

Here the Zeroconf inventor is giving a Google Tech talk, explaining the user experience they have in mind:
https://www.youtube.com/watch?v=ZhtZJ6EsCXo

bubnikv, 2019-03-01:
By the way, Bonjour is not supported on Windows without installing an Apple
driver, which is an annoyance IMHO, which I would refrain from as a user.

Windows supports its own LLMNR protocol. I believe OctoPrint broadcasts
both.

probonopd, 2019-03-01:
You could easily ship your own Bonjour client on Windows as part of the application. I think Cura does this.

bubnikv, 2019-03-01:
Actually I was wrong, our Slic3r implements the Bonjour discovery protocol
on its own. One needs to install the Apple Bonjour service if he wants to
enter the name instead of the IP address into Slic3r's address field, but
that is not the usual work flow. Just for discovery, Slic3r's
implementation is sufficient.

Also if one wants to enter the OctoPrint's Bonjour address into the web
browser by name, one would need to install the Apple driver package.

probonopd, 2019-06-03:
Back to the original topic, which is about improving the UI so that the full potential of Zeroconf (having to do "zero configuration") is utilized.

Today, one has to:

1. First select a printer from the wizard, then 
2. Go to the Printer Settings Tab, then
3. Click "Advanced", then
4. Click "General", then
5. Click "Browse", then
6. a dialog shows up that shows all OctoPrint printers on the network

It should be the other way around:

1. When you launch PrusaSlicer, the dialog shows up automatically that shows all OctoPrint printers on the network, if printers were found
2. When you select a printer, the matching printer profile is automatically selected (based on the recognized firmware)

closed 2026-09-29 by prusaslicer-bot[bot]