prusa3d/PrusaSlicer #1883: Do infill before perimeter if perimeter would be floating in air

haarp, 2019-03-01:
Sometimes it's necessary to have small holes on top of empty space. See these example pictures:

[image]
[image]

Slic3r will try to do the perimeter first, which of course fails as it has nothing to adhere to. It then does the brdiging infill successfully. On the next layer, it starts with the perimeter again, which then successfully adheres to the infill of the previous layer.

It's clear that in such cases, there is absolutely no point in starting with a perimter that floats in air. It would be quite clever if Slic3r could detect this and simply do the infill first. This avoids one layer worth of perimeter junk hanging under your print.

Thank you!

madsi1m, 2019-03-03:
I have also seen this

bubnikv, 2019-03-03:
this is not printable without supports anyway.

haarp, 2019-03-06:
> this is not printable without supports anyway.

Not true, it prints perfectly fine once the infill is set down.

bubnikv, 2019-03-06:
> Not true, it prints perfectly fine once the infill is set down.

What printer? What slicer? I want to see it with my own eyes.

haarp, 2019-03-06:
> > Not true, it prints perfectly fine once the infill is set down.
> 
> What printer? What slicer? I want to see it with my own eyes.

On a normal Prusa Mk3 and and Slic3rPE 1.41.3.

As I said in the initial post, once the infill is set down, subsequent perimeters adhere to that just fine. Only the first perimeter falls through, which is what this issue is about in the first place.

I could record a video if you don't believe me.

rtyr, 2021-01-26:
Related to https://github.com/prusa3d/PrusaSlicer/issues/1860.

DrLex0, 2022-06-21:
> > this is not printable without supports anyway.
> 
> Not true, it prints perfectly fine once the infill is set down.

In this case, it might. But in general, printing infill before perimeters for bridging structures is a horrible idea. Generally it are the perimeters that provide anchor points for the infill, not the other way round.
If there is one thing I have learned over the years, it is not to rely on eventualities and lucky coincidences. One cannot reliably print holes in bridging layers without providing support for the hole perimeters (either through an actual support structure, or a sacrificial layer). I would never gamble that this kind of _trick_ would work in the last few layers of a huge 8-hour print, just to save a bit of material needed for a support, or the bit of extra time needed to drill through a sacrificial solid bridging layer.

Maybe it's getting time for the developers to go through the ever increasing list of open issues, and weed out the ones like these, where some complicated and dubious workaround is requested for what boils down to poor practice. I bet I have also filed a few issues like these in my first few years of 3D printing, it takes a while (and the pain of failed prints) before the necessary experience is gained.

JanNogga, 2024-10-25:
This feature would be very helpful when embedding bearings into prints. I'm sure there would be many other uses cases for people trying to get the most out of their printers.

madsi1m, 2024-10-27:
OR look at https://github.com/supermerill/SuperSlicer/releases/tag/2.5.60.0 at the `overhangs_max_slope` feature

JanNogga, 2024-10-28:
Exactly what I needed, thanks!

https://github.com/user-attachments/assets/827524ef-2ade-4f5a-b510-e88d43010e95

nmschulte, 2025-04-13:
I too have been looking for a solution about bridges where the slicer wants to print the perimeter(s) before the infill; leading to circular perimeters printed afloat and no support.  I've been combating this by clever placement of modifier shapes targeting the area of concern, causing the slicer to create "boundaries" depending on how the modifier varies from the global setting.  However, being unable to directly control the print order of the [modified] areas, I still wish I could print infill before perimeters, especially as a modifier (there is a global setting for this already!).  https://github.com/prusa3d/PrusaSlicer/issues/12914

> In this case, it might. But in general, printing infill before perimeters for bridging structures is a horrible idea. **Generally it are the perimeters that provide anchor points for the infill, not the other way round.**

> This feature would be very helpful when embedding bearings into prints. **I'm sure there would be many other uses for people trying to get the most out of their printers.**

What would be nice and not mentioned yet, is the ability to print the "inner most" perimeter(s) [set] (the smaller, contained/concentered hole ...) last; print the outer perimeter(s) [set] first, then the infill/..., then the inner perimeter(s) [set].

https://github.com/user-attachments/assets/9faf4279-64c7-45d2-908c-9e4399dc4c27

timur-tabi, 2026-09-02:
This is still a useful feature to have.