prusa3d/PrusaSlicer #1886: Slic3rPE not working correctly

handyandyS3, 2019-03-02:
### Version
1.41.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3

### Behavior
I have always put my .stl files to be sliced on the desktop. When I start Slic3r and try to load one of the .stl files it either won't, or it takes ages to show any of the files. I usually run out of patience minimise the program and drop files in that way! It has done this for the last few updates. Also the screen minimise/maximise where you can resize the program does not seem to work either.

bubnikv, 2019-03-11:
> I usually run out of patience minimise the program and drop files in that way!

Are you able to use Slic3r after loading the file?

What graphics card are you using? Would the current Slic3r 1.42.0-alpha7 work better?
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha7

handyandyS3, 2019-03-12:
Hi,
Thanks for the reply. Trying to load a .stl file I place on the desktop 
is still the same - very slow to show any files, at least 15-20 seconds 
at best. It shows the desktop location instantly but initially with no 
files to see?
Once loaded though all seems to work fine. Graphics are excellent on my 
GTX1060 Nvidea graphics card.
One thing - should there be an install procedure for the Slic3r software 
so that it registers file types/locations etc. All I am doing is 
downloading the files from github and then copying the unzipped 
directories into my Prusa subdirectory. I then make a short-cut to the 
slic3r software to run it.
Love the software. For me it is the best slicer I have used (and I’ve 
tried them all!) if only you could get supports similar to Simplify3d or 
Ideamaker it would be near to perfect!
Best regards, Andy

<author@noreply.github.com>
Sent: 11/03/2019 16:33:47

bubnikv, 2019-03-17:
> Thanks for the reply. Trying to load a .stl file I place on the desktop
is still the same - very slow to show any files, at least 15-20 seconds
at best. It shows the desktop location instantly but initially with no
files to see?

Maybe your expectations are too high? How long does it take to open the 20mmbox? By the way, how long does it to load the models for example into Simplify3D? Is it much faster?

[20mmbox.zip](https://github.com/prusa3d/Slic3r/files/2974985/20mmbox.zip)

bubnikv, 2019-05-04:
any news? Would you please answer my questions?

handyandyS3, 2019-05-06:
Hi, I did some Windows 10 maintenance and it all seems ok with the latest version. 
Best regards, Andy

⁣Sent from BlueMail ​

bubnikv, 2019-05-06:
cool, thanks.

closed 2019-05-06 by bubnikv