prusa3d/PrusaSlicer #1889: [Feature Request] top/bottom perimeters overwite

madsi1m, 2019-03-03:
### Version
1.42.0-alpha7

### Behavior
I like to print with 3 to 5 perimeters as it adds a lot of strength. The downside to this is the top and bottom layers look a little ugly as you can see then many perimeters.

Is it possible to add a feature that lets me select how many perimeters I would like on the top and bottom layers separately?

OR

Have an option to drop a perimeter each step, for example, on a 4 perimeter cube, the top layer would be 1 perimeter, next layer would be 2 perimeters, next 3 perimeters then 4 perimeters. And the opposite on the bottom side of the cube. So on a very small cube it would be something like  1,2,3,4,4,4,4.......4,4,4,4,3,2,1 perimeters?

_Is this a new feature request?_
YES

Cheers

BerndJM, 2019-03-03:
Are not exactly for such cases the modifier intended?
To stay at your cube-example:
Simply add a modifier with length & wide of your cube and a height eg 0.4mm
for this modifier you set the perimeters to 1 or two
than you repeat this with a second modifier but this one you move in Z to fit your cubes height ...

madsi1m, 2019-03-03:
I agree, for a cube this is easy.

On objects that are somewhat more complex this is more difficult as there will be multiple top layers.

closed 2026-09-29 by prusaslicer-bot[bot]