prusa3d/PrusaSlicer #1890: [1.42.0 a7] "Reload from disk" non-functional

nebbywan, 2019-03-03:
### Version
1.42.0 alpha7

### Operating system type + version
Win 10 Home v1809 build 17763.316

### Behavior
Selecting "reload from disk" from a part's context menu is not functioning. If a part has been manipulated on the plater, the part in its original orientation can be seen briefly before disappearing. When the same version is run in debug mode, the following message is displayed:
[image]

#### Project File (.3MF) where problem occurs
[Ikea Corner Camera Mount.zip](https://github.com/prusa3d/Slic3r/files/2922758/Ikea.Corner.Camera.Mount.zip)

bubnikv, 2019-03-04:
We are able to reproduce it. Thanks for the bug report.

bubnikv, 2019-03-13:
@vojtechkral

vojtechkral, 2019-03-15:
I'm unable to reproduce the problem on either Linux or Windows.

nebbywan, 2019-03-15:
Still occurring for me on the latest commit (fdf59f7). Also got another debug error when I hit ignore on the one in the original report: 
[image]

bubnikv, 2019-03-17:
I see you are recompiling yourself on Windows.
Would you please attach a call stack? If you click retry, it should open the debugger.
Also which wxWidgets are you compiling against? 3.1.1?

nebbywan, 2019-03-17:
Yeah, I'm building against 3.1.1. Here's the call stack:
[Call Stack.txt](https://github.com/prusa3d/Slic3r/files/2975619/Call.Stack.txt)
And here's the output in case that helps at all, too:
[Output.txt](https://github.com/prusa3d/Slic3r/files/2975622/Output.txt)

bubnikv, 2019-03-17:
thanks, that should help

vojtechkral, 2019-03-18:
@nebbywan Would you please re-test with 456d09be766c3323e652b83952a502ec3b66e313 ?

nebbywan, 2019-03-18:
While commit it fixed the debug assertion, unfortunately "reload from disk" is still non-functional for me. When selecting this action after manipulating a part, the flashing of the original orientation that is seen in the video I posted originally still occurs, and if the action is selected after "cutting" a part, the following error occurs before that half of the part disappears:
[image]

nebbywan, 2019-03-22:
@vojtechkral @bubnikv Just checking in. With commit 4be6c17fd358d19e8fce0766a85b081f9315afe5 I was able to get the debug assertion to sporadically happen. Would you guys prefer I submit a new ticket or would you rather reopen this one? Just in case, here is the the call stack and output logs.
[Call Stack.txt](https://github.com/prusa3d/Slic3r/files/2995387/Call.Stack.txt)
[Output.txt](https://github.com/prusa3d/Slic3r/files/2995385/Output.txt)

bubnikv, 2019-03-24:
If you believe it is still broken on the git master, then please reopen this issue.

nebbywan, 2019-03-24:
@bubnikv could you reopen this then, please? Since a collaborator closed it I don't have permissions to do it myself.

nebbywan, 2019-05-31:
Sorry for not checking again sooner, but the issue seems to be resolved. Closing.

closed 2019-05-31 by nebbywan

biskero, 2019-06-07:
I have this issue.
I updated an stl and than tried to "reload from disk" but old stl is still showing even though is not there anymore since it has overwritten from new one.

TickTock-, 2019-06-17:
Same.  Reload from disk works as long as I don't save a project.  If I just place parts without saving a project I am able to reload an updated STL but I have never been able to load a saved project and re-load from disk to get the updated STL (project comes up with the old STL even though the file has been replaced on the drive - I guess it is stored in the 3mf file?).  Maybe the project points to its own internal copy and forgets the original location so it just reloads its internal copy?

TickTock-, 2019-06-17:
I guess the STL isn't stored in the .3mf since that file is smaller than the STL.  So the tool must be saving a shadow copy somewhere.  This is very repeatable.  Even cycling power on the PC will result in the project loading with a an STL model that no longer exists and it does it with any model.  No umlaut.  My current test case is just Body1.stl.  I tried putting the STL in the same directory as the .3mf but that made no difference.   Where does PrusaSlicer keep it's local copies?

commit 456d09be (2019-03-18): Fix out of bounds selection in ObjectList, fix #1890