prusa3d/PrusaSlicer #1891: Print Settings get erased if not saved but another printer is selected.

MartinMajewski, 2019-03-03:
### Version
1.42.0-alpha7

### Operating system type + version
MacOS Mojave 10.14.3

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2.5

### Behavior
1. Change something in the Print Settings without saving the profile ("modified" is stated behind profile name)
2. Switch Printer (Select another printer)
3. Print Settings' changes get erased - selected profile is restored!

This is a very annoying behavior because often one wants to adjust a setting quickly and start another job on a different machine. But if one does not select the correct machine beforehand, the setting changes get erased. Changing every setting into a new profile or overriding an existing profile is not always wanted; maybe the settings are just for testing or the changes are minor, and one does not want to pollute the settings menu with setting profiles.

bubnikv, 2019-03-04:
We are able to reproduce it. Thanks for the bug report.

bubnikv, 2019-03-04:
This should be fixed with 6f0efd11688fff890be75e9000e7200180a2c7af

MartinMajewski, 2019-03-04:
As I don't have time to go through all build step hurdles I will patiently wait for the next official test build for Mac. Thank you, @bubnikv !

bubnikv, 2019-03-11:
You should thank to @YuSanka who fixed it.

MartinMajewski, 2019-03-11:
@YuSanka 🙏 Thanks a lot!

bubnikv, 2019-03-17:
It will be fixed in the beta. Closing.

closed 2019-03-17 by bubnikv