prusa3d/PrusaSlicer #1892: Keep track of color changes relative to object base

sgoll, 2019-03-03:
### Version
1.42.0-alpha7+linux64

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any

### Feature request
In the current alpha version, the ability to add color changes was added. The layer height for changes can be specified in the Plater.

I'd suggest keeping the selected layer height constant within the object when changing print settings that add additional layers such as the raft option. This is already true when changing to a different layer height (i.e. the absolute height in millimeters is kept the same, i.e. the color change may now occur on a different layer number).

When adding or removing raft layers, however, any previously configured color changes now move across the object because the _absolute_ height of the color change is kept the same, however, arguably the more common use case would be to fix the height within the object.

In other words, when configuring a color change 5 millimeters into the object, this should not change when later adding raft layers underneath the object: the color change should still occur 5 millimeters into the object, which may now be at a total build height of, say, 7 millimeters (because of the additional raft layers).

Internally, negative numbers could be used if a color or material change is actually required within the raft layers (`0` being the base of the object). Having a raft of 3 millimeters, a color change at height -1 millimeter would mean an absolute print height of 2 millimeters into the raft.

What do you think?

bubnikv, 2019-03-04:
You are right, that the color change shall be referenced to the bottom of
the object, not to the print bed. It is certainly a bug.

YuSanka, 2019-03-07:
Working on it in [SPE-848](http://dev.prusa3d.com/browse/SPE-848)

bubnikv, 2019-03-11:
Your request is difficult to fulfill in case where you enable the raft for one object, but disable for another one.

With this in mind, I would propose to keep the behavior as it is, as it is consistent at least to some extent.

sgoll, 2019-03-13:
@bubnikv You mean because of the single layer selector bar? If some objects have rafts and some don't, I guess the bar would have to change labels depending on the current object selection.

bubnikv, 2019-03-13:
>   If some objects have rafts and some don't, I guess the bar would have
to change labels depending on the current object selection.

But the color change applies to all objects, not just the object selected.

sgoll, 2019-03-13:
@bubnikv I see: this kind of simple color change is indeed dependent on the absolute layer level. In that case I have to retract my feature request/suggestion. For anything that is more complex (such as having different color changes _per object_), you'd basically have to manually change filaments after every layer (until a layer uses only a single tool again) which seems hugely impractical without a machine doing it for you, such as the MMU.

TL;DR: I was mistaken, this simple color change should indeed be given in absolute layers and not in object dimensions.

closed 2019-03-13 by sgoll

bubnikv, 2019-03-13:
> I was mistaken, this simple color change should indeed be given in
absolute layers and not in object dimensions.

I would not say that you were mistaken, you raised a valid objection. Only
that there is not a clean resolution, so I would rather keep the existing
simple solution in place.

Thanks for understanding.