prusa3d/PrusaSlicer #1893: When scrolling side panel on OS X, dropdown changes when passing under cursor.

shendrick, 2019-03-04:
### Version
_Version of Slic3r Prusa Edition used goes here_
Slic3rPE-1.42.0-alpha7+full-201903011807

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #1325 
macOS Mojave 10.14.2
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
When scrolling the panel using the touchpad, if you scroll with the mouse over a dropdown such as the filament selection dropdown, this scrolls through the filaments and can go unnoticed if you aren't careful.
* _Steps needed to reproduce the problem_
See above.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
The dropdown selection shouldn't change when scrolling over with touchpad.
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

FidelCapo, 2019-03-19:
Thank you for report.
Created JIRA-876

bubnikv, 2019-03-19:
Thanks for the report. We will try to disable scrolling over the combo boxes by the mouse wheel or by the gesture on all platforms.

bubnikv, 2020-12-12:
Fixed in PrusaSlicer 2.3.0-beta2.

closed 2020-12-12 by bubnikv