prusa3d/PrusaSlicer #1894: MMU idler Movement

UlwaziSA, 2019-03-04:
### Version
_Version of Slic3r Prusa Edition used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

UlwaziSA, 2019-03-04:
My idler on the MMU moves left to right or right to left even when there is filament in. Should the Pinda not trigger a fail safe to ensure idler can not move while filament is in ?

bubnikv, 2019-03-04:
This sounds like a firmware or mechanical problem.
If you believe it is a firmware problem, please report it here:
https://github.com/prusa3d/prusa-firmware

closed 2019-03-04 by bubnikv

UlwaziSA, 2019-03-04:
I reported on the Sklic3r help button. It took me to the page