prusa3d/PrusaSlicer #1896: skirt smaller than object

guestisp, 2019-03-04:
I've found this: when printing a complex object, skirt is generated smaller and badly positioned, the object is printed above the skirt.

[image]

bubnikv, 2019-03-04:
Please provide a 3MF.

guestisp, 2019-03-04:
I don't have a 3MF but the object is from thingiverse: https://www.thingiverse.com/thing:1426081

BerndJM, 2019-03-04:
If you have a close look at the model you will see that the bottom isn't absolute flat.
So independend from the skirt, you will print some layers in "free air".
Just cut it in Z at 0.8mm and all will be fine ...

guestisp, 2019-03-05:
Not flat? I've printed that object yesterday and didn't noticed anything strange with the bottom

bubnikv, 2019-03-17:
The tail has a tiny bump. It looks like a 3D scan, which was not processed properly.

[image]

To fix it, you can use the cut tool to trim the bump away.

[image]

[image]

Everything works as intended, closing.

closed 2019-03-17 by bubnikv