prusa3d/PrusaSlicer #1897: Crash when changing the number of perimiters in a modify area

craftit, 2019-03-04:
### Version
1.43.3+

### Operating system type + version
MacOS Mojave 10.14.3

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3

### Behavior

I have a part which I want to reinforce in some areas.   To achieve this I create a modifier which defines which areas of the part need to be stronger.   When I try and set the number of perimeters on the modified area slic3r crashes.

1) Load the attached file.
2) Goto 'settings' for the object
3) Select BarrelBearingMod.stl 
4) Use the green + to add a modifier 'Layers and Perimeters'  and then 'Perimeters'
5) Enter a new value, I used 8.  
6) Hit return or ok, Slic3r then segmentation faults.
 
#### Project File (.3MF) where problem occurs

[barrel.zip](https://github.com/prusa3d/Slic3r/files/2928437/barrel.zip)

bubnikv, 2019-03-17:
We are going to release the 1.42.0 beta soon. Would you please retest with the beta?
https://github.com/prusa3d/slic3r/releases
I quickly tested and it seems to be working correctly (at least I don't get the crash).

The 1.41.3 will not be fixed.

craftit, 2019-03-17:
I just tried it, and it does look like it is working correctly.  The new version is looking great too! Thanks!

bubnikv, 2019-03-17:
great, closing.

closed 2019-03-17 by bubnikv