prusa3d/PrusaSlicer #1899: [Feature Request] Add a Pause Print at a Specific Layer Height

CRKatri, 2019-03-05:
### Version
1.42.0-Alpha7

### Behavior
I would like a button to add a pause at a specific layer height for embedding objects in 3d prints, possibly a button next to the button that adds a color change

_Is this a new feature request?_
Yes

BirkBinnard, 2019-03-05:
The firmware for my Atom2 printer includes a Pause/Move and a Resume command. The Move part raises the printhead about 4 cm above it's current position; this makes it easy to change filament colors or replace a near empty reel with a new one. THe Resume command lowers the printhead to where it was and simply resumes printing.

CRKatri, 2019-03-05:
I was thinking it would pause and move the head up and the bed forward then wait for you to resume the print through the lcd

vlcty, 2019-03-10:
I think this feature would be really cool. As a workaround you could use the filament change code M600. It's pausing and moving the print head away waiting for you to change filament. The odd thing is that you have to unplug the filament and shove it back it but it's the simplest thing you can do.

Either use the Prusa tool for multicolor print: https://www.prusaprinters.org/color-print/
Or add a custom G-Code after layer change:

```
{if layer_num == youlayernumhere}
M600
{endif}
```

BirkBinnard, 2019-03-10:
Thanks victy, I've never checked out the M600 command. I don't have a Prusa printer so I don't know how to get the printer to start working again. Maybe the explanation for the M600 explains how to do this.

vlcty, 2019-03-10:
Yeah it's done by the printer firmware. On Prusa printers it lifts the nozzle up a bit and moves the extruder to the front. The printer guides you through the filament change. I hope your printer supports it! :-)

bubnikv, 2019-03-17:
We are discussing this feature with the firmware team. @YuSanka will possibly implement this feature into Slic3r.

BirkBinnard, 2019-03-17:
Excellent news - thanks. This capability would make it possible to change filament colors every 200 mm (for example) of height throughout the print. Very spiffy indeed.

DeanDavis, 2019-05-16:
Please see this for good technique for introducing a pause in the print for embedding magnets or nuts.
[https://www.youtube.com/watch?v=oF1SdIR-Kow](url)

Being able to do this in PrusaSlicer using the user interface rather than modifying the GCODE manually would be awesome.

ggrote, 2019-05-23:
G1 X10.000 Y200.000 E0; parking position
M1; user stop
M105; return to current temp

works best for adding magnets, because the printplate is moved to the front and no filament change is required. you can continue the print with a push on the button at the printer. or if you print via octoprint just click continue in the web interface.

DeanDavis, 2019-05-23:
Shame, looks like there was a setting in PrusaSlicer 2.0.0+ in "Printer Setting" in the Custom G-Code section that would allow us to put this code in as a temporary workaround.
If this section was usable then you could put the "Pause" GCode in there.
But  I can't seem to edit this section.

icanhazpython, 2019-09-01:
I would love to see this as well. I currently do a find/replace in a text editor to replace M600 with the following:
```
G91                  ; Put in relative mode
G1 Z10               ; Lower bed by 10mm
G90                  ; Put back in absolute mode
G1 X0 Y0             ; Zero (home) the X & Y
M300 S300 P1000      ; play beep sound to signal attention
M0 Click To Restart  ; Pause and wait for the user
G91                  ; Put in relative mode
G1 Z-10              ; Raise the bed back up 10mm
G90                  ; Put back in absolute mode
```
The ability to specify this custom code instead of M600 for color change layers would be ideal. [Adding M125 support to the firmware](https://github.com/prusa3d/Prusa-Firmware/issues/2110) may be a further improvement.

vlcty, 2019-09-01:
PrusaSlicer has now the ability to add a color change at a specific height.

ex-nerd, 2019-09-01:
@vlcty Is this a new feature for multimaterial printers?  I believe the reason this ticket was opened is because the feature only shows up for single-material printers.  It's just that some of us also use the feature to pause the print at certain heights (e.g. to insert an embedded nut) rather than to actually change colors.

vlcty, 2019-09-01:
Oh yeah. I'm so sorry! I somehow thought this topic was about changeing filamet at a specific height! I don't know any new feature other than filament change to pause at a specifiy height.   
Well actually: You could do a pseudo filament change of course. So you unload the current filament and then load it again.
Again sorry for the disturbance!

Sembazuru, 2019-09-19:
I use a version that I've modified from a post on the forum by Joan Tabb (link in the gcode comments).

```
{if layer_z=4.74}; Pause to insert objects
M400 ; wait for movement buffer to empty
M300 ; beep
M104 S0; Turn off extruder heater to avoid too much oozing
; code snippet originally Prusa forums joantabb here: https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/pause-print-in-gcode/#post-125758
G1 X10.000 Y200.000 E0; parking position GET THE PRUSA TO STICK IT'S TONGUE OUT! (edited to revise Y coordinate)
M400; wait for bed to stop moving
M300; beep again
M84; disable motors (may not be necessary)
M0; user stop
G28 X Y; Home X and Y (only) to correct for accidentally moved bed or extruder (it is assumed Z didn't move because that is a harder axis to manually move).
M109 S[temperature]; return nozzle to printing temperature
{endif}

```
Change the layer_z value to what is required for your print, copy the full if...endif statement as many times as you need to pause and enter the proper layer_z value for each copy.

An error above (and in the forum thread) is using M105 to turn the hot end heater back on. M105 only returns the temperatures to the console and M0 doesn't seem to turn off either the hot end nor the bed heaters. Turning off the hot end reduces oozing while paused but the bed stays hot so the part doesn't pop off of a cooling bed.

ex-nerd, 2019-10-03:
@Sembazuru Where exactly do you put that custom gcode?  I assume it's one of either "before" or "after" layer change.

Sembazuru, 2019-10-03:
@ex-nerd I put the custom code in the "before layer change" g-code. Good luck with your prints.

ex-nerd, 2019-10-05:
I noticed that several of the gcode solutions posted above were generating errors in the current prusaslicer release.  The comparison operator should be `==` not `=`.  And since my Duet-based printer supports it, the pause command should work just fine with `M600`:

```
; For reference: https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language
; Pause to insert objects (e.g. a nut or magnet)
{if layer_num == 58}
M600
{endif}
```

James-G4CLF, 2019-11-16:
Remember Murphy's Law*. The problem with manual Gcode edits is the possibility of mistakes. Putting in an "Insert Object" function, once it's debugged, is a really good idea. Using "Change Filament" does not (at present anyway) leave easy access for inserting nut, magnets, blades or things to be smuggled, and replacing the ejected filament is not only a (very minor) inconvenience but also an additional opportunity to accidentally do something damaging to the print.

* "If it can go wrong, it will go wrong."

macropin, 2019-12-20:
A feature teaser from Josef https://twitter.com/josefprusa/status/1208063594950266881. Looks like we'll see this soon.

macropin, 2019-12-21:
Alpha build that adds this feature is available https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha1

nordurljosahvida, 2019-12-29:
In my gcode generated with 2.2.0-alpha2 for `MK3SMMU2S Single mode` the code inserted is `M601` which leads the printer into an [infinite pause/resume loop](https://github.com/prusa3d/Prusa-Firmware/issues/1378). Is there a reason `M601` is being used rather than `M600`?

ModischFabrications, 2020-01-05:
My CR-10S simply ignored the M601, is a specific version (of marlin?) needed?

neophyl, 2020-01-05:
Its not versions of Marlin, it depends on if the Creality implementation of Marlin supports M601.  Not all vendors support all the possible codes.  The Prusa firmware only supports a subset too.

icanhazpython, 2020-02-04:
I'm currently testing the new "insert pause" implementation in slicer 2.2.0 alpha 2. I think one nice thing to have would be a time estimate to each pause just as you get when you insert color changes.

bubnikv, 2020-02-05:
The feature asked for has been implemented in PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-02-05 by bubnikv

zalexzperez, 2020-03-31:
What if I want to change the color of an engraved text only at the exposed text area and NOT at the whole layer as it happens for a typical filament change?

What I'd like to do is, place the face with the text touching the bed (take Prusa LCD case as an example), so that the color change only takes place for the bridge infill. However this forces you to have a single layer for this color.

I can't also find a way to print this in 3 parts, as printing something on top of something that's already printed doesn't seem to be a possibility at the moment?