prusa3d/PrusaSlicer #1900: Change of settings not synchronized between "Print Settings" tab and "Plater" tab

rschoem, 2019-03-05:
### Version
1.41.3 Alpha 7

### Operating system type + version
Windows 10, 64 bits

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3

### Behavior
When I change a setting in the "Print Settings" tab (e.g. layer height) and switch back to the "Plater" tab, the changed value is not reflected in the "Object Settings to Modify" panel (i.e. old setting is still shown).
Also, if I select "Slice Now" the unchanged value (from the "Object Settings to Modify" panel) seems to be used for slicing.

Note: the other way around seems to be working correctly: if I change a setting in the "Object Settings to Modify" panel, it is correctly shown in the "Print Settings" tab.

bubnikv, 2019-03-17:
We are going to release the 1.42.0 beta soon. Would you please retest with the beta?
https://github.com/prusa3d/slic3r/releases

The 1.41.3 will not be fixed.

bubnikv, 2019-03-17:
You noted the Slic3r version as 1.41.3 Alpha 7, I think you meant 1.42.0-alpha7.
I just re-tested with the upcoming 1.42.0-beta and it seems to be working.
Closing. If you still have the issue, please reopen.

rschoem, 2019-03-17:
I tested with 1.42.0 beta (Mac this time). Issue still is present.

bubnikv, 2019-03-18:
it would help then to provide more details. A video would help.

bubnikv, 2019-03-18:
@rschoem: I tested with 1.42.0 beta (Mac this time). Issue still is present.

bubnikv, 2019-03-18:
I believe there is a misunderstanding of the purpose of the "Object Settings to Modify" panel. These values are object specific overrides. These object specific overrides are once initialized from the profile (print profile / filament profile / printer profile), but then they live on their own. Their sole purpose is to override the global parameters (those selected by the print / filament / printer profiles) for a single object.

rschoem, 2019-03-18:
Ah, ok, in that case I misunderstood the purpose of the panel. Sorry for the fuzz then :-P

Regards Rob

bubnikv, 2019-03-18:
Great, closing.

closed 2019-03-18 by bubnikv