prusa3d/PrusaSlicer #1901: Sequential printing of objects in different colors does not wipe the previous filament

chrschn, 2019-03-05:
### Version
Version 1.41.3+linux64

### Operating system type + version
Gentoo Linux

### 3D printer brand / version + firmware version (if known)
Prusa MK3 + MMU2.0

### Behavior
* Printing multiple uni-color objects in different colors using the output option "Complete individual objects" and selecting different tools for the objects does not wipe the color after a tool change.
* _Steps needed to reproduce the problem_
  * Add two objects to the platter
  * Assign two different tools for the objects
  * Check the box in Print settings -> Output options -> Complete individual objects
  * Export the GCode
* _Expected Results_
  * Slic3r should create a wipe tower for the tool changes
* _Actual Results_
  * No wipe tower is created. As a result the tool change between the objects happens, but the filament is not extruded in the correct color.
  [image]
_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
[wipe_tower_30mm3_0.2mm_PLA_MK3MMU2.3mf.gz](https://github.com/prusa3d/Slic3r/files/2930939/wipe_tower_30mm3_0.2mm_PLA_MK3MMU2.3mf.gz)

bubnikv, 2019-03-17:
Wipe tower is not supported for sequential printing.

tobymurray, 2019-05-24:
Is this a duplicate of #1787

bubnikv, 2019-06-03:
Yes. Closing this one.

closed 2019-06-03 by bubnikv