prusa3d/PrusaSlicer #1903: error message when saving a preset

cybergogo, 2019-03-05:
Hello,

first, here are my versions info:
Slic3r Prusa Edition
Version: 1.41.3+win64
Build: Slic3rPE-1.41.3+win64-full-201902111723
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 10.0.17763.316] 

Here is my issue : when I edit a print or filament setting and try so save it, the first try shows an error message saying that i can't overwrite a system preset.
If I try again, it shows the normal dialog box asking for a preset name to be saved.

Don't know if it's a bug or intended in a weird way. 
Just wanted to share the issue ;-)

supermerill, 2019-03-07:
If the preset has a lock at his left, it's a system preset and they can't be erased.
When you save a preset, if the name already exist, it delete it and replace it by the new one.
So, for saving an edited system preset (the ones with the lock), yo have to change the name when saving.

cybergogo, 2019-03-07:
> If the preset has a lock at his left, it's a system preset and they can't be erased.
> When you save a preset, if the name already exist, it delete it and replace it by the new one.
> So, for saving an edited system preset (the ones with the lock), yo have to change the name when saving.

That's the thing, the first attempt to save the preset after I install slic3r on my pc, it doesn't show the "save preset" window where i'm supposed to enter the new name, it only shows an error message saying i can't overwrite a preset blablabla...

After that, i try again with any system preset, and when i click on save, it displays the "save preset" window directly. It should have done that from the start.

bubnikv, 2019-04-13:
From the change log of https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

> When saving a system profile, a suffix "- Copy" is added to the name of the system profile to avoid a name clash with the system profile.

I believe that covers your issue. Closing.

closed 2019-04-13 by bubnikv