prusa3d/PrusaSlicer #1904: Missing layers when slicing

guestisp, 2019-03-05:
Take the queen from this (but even other pieces): https://www.thingiverse.com/thing:40605
Layer 311 (layer height 0.2, first layer 0.3) is too far from the object, the nozzle will extrude in the air.

I've checked the same STL, same settings with cura and slicing is ok (and so does the print)
I think there is a bug somewhere, probably an additiona layer between 310 and 311 should be added.

guestisp, 2019-03-05:
This happens only with the alpha version.
The stable one, for the same object, make much more layers (469 in the alpha7, 781 in 1.41.3)

The same is with a 3DBenchy, 239 layers with the alpha, 399 with 1.41.3

guestisp, 2019-03-05:
I've tried all alpha releases. All affected. With all releases, there are missing layers.

vojtechkral, 2019-03-05:
(Issue accidentally closed?)

guestisp, 2019-03-05:
Closed for a reason but for another reason i came back here to reopen

Please ignore layer number between alpha and stable, i had two different settings, but the original issue is real, cura does a better slicing in this case, with no "flying" layers

The queen can't be printed with layer 0.2 with slic3r while it's printed perfectly with the same layer height in cura (v3.6 and v4)

guestisp, 2019-03-06:
Just as confirm for me: did you see the mentioned issue or you just added the tag ?

bubnikv, 2019-03-11:
Would you please attach a 3mf, so I can retest?

guestisp, 2019-03-12:
I've attached a 3mf file
[Queen.zip](https://github.com/prusa3d/Slic3r/files/2956057/Queen.zip)

FidelCapo, 2019-03-13:
Unfortunately, I am unable to reproduce this issue. Can you please attach screenshot of missing layers? Thank you.

guestisp, 2019-03-13:
They are not "missing", they are created too far from the previous layer, thus extruding mid-air.
Same identical settings with Cura doesn't create this issue and extrusion is done properly.

I've thought as thinking layers because adding an additional layer just before the first blu line (overhang) could reduce the distance and fix the issue.

[image]

bubnikv, 2019-03-13:
I believe this is perfectly correct. You have the "detect bridging perimeters" feature enabled, which extrudes a bridging flow where the overhangs are too steep to be supported by the underlying layers. You may try to print the piece with and without the bridging perimeters and compare, whether the bridging perimeters were helpful or harmful.

guestisp, 2019-03-13:
By removing the `detect bridging perimeters` blue lines were removed, not the layer preview seems to be correct. So, I think there is something strange in detect bridging perimeters feature, if Slic3r add a un-needed bridging perimeter in a way that can't be printed. Would be ok by adding a bridging perimeter, but it must be placed near the underlaying layers, this is too far away. What's the meaning of extruding mid-air ?

bubnikv, 2019-03-13:
> So, I think there is something strange in detect bridging perimeters
feature, if Slic3r add a un-needed bridging perimeter in a way that can't
be printed.

If you think that it cannot be printed, then you should enable supports.

guestisp, 2019-03-13:
Sure, but with the same identica settings, cura print this properly without support (this piece is desined to be printed without supports). I have this issue only with slic3r

But my question is legit: why adding a bridging perimeter that can't be printed while the same piece without briding perimeter is being printed properly ? Currently, in this case, briding perimeters will add a unprintable layer.

bubnikv, 2019-03-13:
> Currently, in this case, briding perimeters will add a unprintable layer.

Are you sure that the layer is unprintable? How about to print the two pieces, with / without the bridging perimeters and to compare the effects? Maybe you will find out that Slic3r prints this part better than Cura.

The bridging perimeters are there to increase chance of printing sloping surfaces with very low angle. If you don't like it, you may disable it. There is no such feature in Cura, that's why you don't see such effects there.

guestisp, 2019-03-13:
> Are you sure that the layer is unprintable?

Sure, i've printed with Slic3r (the slicer i usually use) and there where a mid-air extrusion. Then i've reprinted the same, with same settings with Cura, and extrusion was made properly.

> How about to print the two pieces, with / without the bridging perimeters and to compare the effects? 
> Maybe you will find out that Slic3r prints this part better than Cura.

I'll do in next few weeks as I have to print the second color chess-set

> The bridging perimeters are there to increase chance of printing sloping surfaces with very low angle. 
> If you don't like it, you may disable it. 
> There is no such feature in Cura, that's why you don't see such effects there.

Ok, so Cura is working like Slic3r without the bridging perimeters. Good to know.
But my question is still the same: why slic3r adds an unprintable bridging perimeter? It's too far from the previous layer. That's why i've opened the issue here. I wasn't aware of such feature so I've thought about a missing layer during the slicing, but probably the real reason is slic3r that adds a briding perimeter too far away.

FidelCapo, 2019-03-14:
I did test prints of your object and here is result.
The object on the photo on the left is printed with Detect bridging perimeters enabled and on the right disabled.

[image]
[image]

guestisp, 2019-03-14:
There are mid-air extrusion in your printer, look at layer 311, you can see the filament extruded badly and badly attaching to the print. Cura prints properly.

bubnikv, 2019-03-14:
> There are mid-air extrusion in your printer, look at layer 311, you can see the filament extruded badly and badly attaching to the print. Cura prints properly.

I will leave this issue open for some time, maybe someone will chime in and support your opinion.

IMHO it works correctly as intended. If nobody will support your opinion, I am going to close this issue.

guestisp, 2019-03-14:
These lines should not be present. This is happening because is printing too far from the piece, thus the extruded plastic doesn't attach properly to the lower layer. With Cura i don't have this issue, probably Cura is creating a smoother overhang with multiple layers where Slic3r "skip" one. If you add one layer more between 310 and 311, the overhang will be much smoother and i'm almost sure will fix the print.

[image]

FidelCapo, 2019-03-14:
Result of printing this object using cura
[image]

guestisp, 2019-03-14:
Strange, I did the print with cura and it was printed properly

bubnikv, 2019-03-17:
Works as intended, closing.

closed 2019-03-17 by bubnikv