prusa3d/PrusaSlicer #1905: Multiple retractions

guestisp, 2019-03-05:
Maybe an issue is not the proper place where to ask this but: how can i define different retraction settings (or different settings at all) on multiple objects in the same print?

I've seen that slic3r has multiple threads support like S3D but i didn't see where to set different retraction settings (length, speed, ...) for each object

bubnikv, 2019-03-06:
This is not supported. Retraction settings could only be defined for the
whole print.

guestisp, 2019-03-06:
it's a shame, I have to test retractions and currently i have to make tens of prints :(

Any plans to add this very useful feature ?

bubnikv, 2019-03-17:
Yes, it is a very great shame.

You can implement your test prints using firmware retraction, if your printer supports it.

guestisp, 2019-03-17:
Even firmware retraction are for the whole print

bubnikv, 2019-03-17:
But you can change the parameters of the firmware retraction through the
G-code using the between objects G-code custom section or with a
post-processing script.

guestisp, 2019-03-17:
I wasn't aware of this. Thank you

closed 2019-04-13 by bubnikv