prusa3d/PrusaSlicer #1906: GUI icons too large

wavexx, 2019-03-05:
### Version

Master as of today: c1a24d2fbb78cd5695f9e0b34f3a1cf0688e558c

### Operating system type + version
Debian unstable + Intel UHD (kabylake GT2)

### Behavior
I semi-regularly rebuild slic3r from master. In the last week the OSD icons doubled in size.
Screenshot:

[image]

The current size really wastes too much space.
How is the scaling calculated?

I have two screens attached. They obviously have different resolutions.
I manually set both screens to 95 dpi via xrandr to ensure no GUI scaling takes place.
I also set the Xft.dpi resource:

$ xrdb -q |grep -i dpi
Xft.dpi:        95

The font sizes you see in the GUI are perfect. I would assume the GUI elements are scaled according to that too.

bubnikv, 2019-03-06:
Exactly, the icons are scaled relative to the font size.

wavexx, 2019-03-06:
After looking at this, it seems that toolbals/gizmos are scaled according to wxGLCanvas::GetContentScaleFactor. See

https://github.com/prusa3d/Slic3r/blob/2a6de99f06404b9599f178024d00c239934887b4/src/slic3r/GUI/GLCanvas3D.cpp#L6560

Changing this to ``Slic3r::GUI::wxGetApp().em_unit()* 0.05`` as done elsewhere actually makes scaling work as intended. I've actually seen this change done in several other spots (Tab.cpp uses em_unit() with the old GetContentScaleFactor commented out).

nebbywan, 2019-03-09:
Commit 2daa898 fixed this for me.

bubnikv, 2019-03-09:
@wavexx  are you happy now as well?

wavexx, 2019-03-09:
On Sat, Mar 09 2019, Zach McDonough wrote:
> Commit 2daa898 fixed this for me.

You mean "Top toolbar icon size set to 40px" (172daa898)?

It's ok, but the gizmos on the left and the bottom-left have the same
issue.

And now that I made the connection, when I start slic3r the bed scale is
also too small.

I suppose there's something wrong with the underlying
GetContentScaleFactor.

wavexx, 2019-03-09:
Here's the default/initial startup scale of the bed: 

[image]

nebbywan, 2019-03-09:
Oh wow, can't say I've seen that bed scale weirdness on either my HiDPI machine or the "normal" one. But yes, that's the commit I was referencing. As for the side widgets, I honestly thought they were always in that proportion to the top controls, but I'd have to go back to a pre-HiDPI version to be sure so I'm probably wrong on that one 😉

wavexx, 2019-03-09:
While I was looking at the code, most of the toolbars and icons reference m_canvas->GetContentScaleFactor for the non-apple version. It's done in quite a few places.

It wouldn't surprise me if the initial bed scale would be affected by GetContentScaleFactor as well.

I actually noticed the moment the bed became too small on start, but since I build from master I didn't give too much weight to it.

wavexx, 2019-03-09:
So the small bed scaling on startup happens only when the DPI is of the screen is 200+.
When I run at 95dpi the initial scale seems correct, 

One thing of note is that I'm on debian unstable, which is still using wx 3.0.4.
I know the official alpha is building wx 3.1. Since building also wx 3.1 is annoying, can I ask if it does change anything involving DPI management?

bubnikv, 2019-03-17:
> @wavexx I know the official alpha is building wx 3.1. Since building also wx 3.1 is annoying, can I ask if it does change anything involving DPI management?

Building and testing Slic3r PE for wxWidgets 3.0 and 3.1 is certainly time costly and annoying to us. It would certainly help us, if you can test Slic3r on your HiDPI setup with Slic3r compiled against wxWidgets 3.0. Thank you.

bubnikv, 2019-03-17:
@enricoturri1966 would you please verify these HiDPI issues, mainly the down scaled bed on start up? Thank you.

wavexx, 2019-03-17:
I can confirm the toolbars being oversized on both wx 3.0 and 3.1.2 (aside from the top one, which has fixed to 40px in height).

As for the bed scaling, I have some more news. I don't think it's an hi-dpi issue. By starting slic3r-gui several times in a row, I had several occasions where the bed was scaled correctly on start. Incidentally, the scaling seems to be right if I get an assertion error before slic3r displays the main window: I think the scale is calculated probably too early, before the window gets the final window size.

enricoturri1966, 2019-03-18:
@wavexx: could you please check if https://github.com/prusa3d/Slic3r/commit/710bb66dfc39dad6e13a45329bce4de7acf18c7d helps in solving the "small bed" issue ?

wavexx, 2019-03-18:
On Mon, Mar 18 2019, enricoturri1966 wrote:
> @wavexx: could you please check if 710bb66 helps in solving the "small bed" issue ?

I rebuilt it and so far I didn't see it happen again. Tested in both hi
and regular dpi modes.

bubnikv, 2019-04-13:
Can we close this issue, or has anybody any objection? If so, please provide screenshots to make things clear. Reading the conversation, I am not sure what one considers to be a bug.

wavexx, 2019-04-13:
On Sat, Apr 13 2019, bubnikv wrote:
> Can we close this issue, or has anybody any objection? If so, please
> provide screenshots to make things clear. Reading the conversation, I
> am not sure what one considers to be a bug.

I think the icons are really too big in several cases.

The top gizmos have been "fixed" by pinning the size, which is a bit
odd, but left all the others with a different size.

I was thinking to propagate the changes myself to the other gizmos, but
the code is _quite_ messy.

Why not mock GetScaleContentFactor on x11/win so that it provides the
value the code expects, instead of #ifdef'ing the code in 200 places?

This would get rid of most __APPLE__ ifdefs around retina code and all
the repeated/broken scaling factors elsewhere in the code.

wavexx, 2019-04-13:
Just uploading as requested:

[image]

note how the left/bottom-left gizmos are oversized compared to the top toolbar. I would at least expect both to have the same size as the top.

bubnikv, 2019-04-13:
>  note how the left/bottom-left gizmos are oversized compared to the top
toolbar. I would at least expect both to have the same size as the top.

This is on purpose, and the size of the icons is derived from the size of
the fonts, which you have set quite large.

I think we will HAVE to make the sizes configurable, because someone will
always complain.

> Why not mock GetScaleContentFactor on x11/win so that it provides the
value the code expects, instead of #ifdef'ing the code in 200 places?

Well, we are still fighting with this HiDPI hell. Every platform handles it
differently. Once we have it running reasonably, we may refactor, or better
push it into the wxWidgets.

While we have 3 programmers on our team running Linux, none of them could
give us a definitive answer, how the HiDPI support works on various older
and newer Linux distros with GTK2 vs GTK3 vs wxWidgets 3.0 vs wxWidgets
3.1/3.2/3.3. We will be very happy if you can help us.

> This would get rid of most __APPLE__ ifdefs around retina code and all
the repeated/broken scaling factors elsewhere in the code.

You likely overlooked, that the retina behaves differently from the other
platforms: On Windows and OSX, the UI code and OpenGL code uses the same
coordinate space. On OSX, the UI code uses logical coordinates, while the
OpenGL space uses the physical coordinates. What a mess.

wavexx, 2019-04-13:
On Sat, Apr 13 2019, bubnikv wrote:
>>  note how the left/bottom-left gizmos are oversized compared to the top
> toolbar. I would at least expect both to have the same size as the
> top.
>
> This is on purpose

Ok, this is something I definitely _didn't_ expect.

But can you do a screenshot on your platform and show me what you
consider the sizes as "fine"? I don't have any reference here.

> While we have 3 programmers on our team running Linux, none of them
> could give us a definitive answer, how the HiDPI support works on
> various older and newer Linux distros with GTK2 vs GTK3 vs wxWidgets
> 3.0 vs wxWidgets 3.1/3.2/3.3. We will be very happy if you can help
> us.

I used wx2 with GTK2 significantly 3-4 years ago, but since then
switched to Qt for all newer development purposes (mostly due to the
API: Qt is just better). But nowadays I'm not even sure what's the
direction the GTK team is taking: GTK3 has been an utter usability
nightmare.

> You likely overlooked, that the retina behaves differently from the
> other platforms: On Windows and OSX, the UI code and OpenGL code uses

I assume windows and linux here

> the same coordinate space. On OSX, the UI code uses logical
> coordinates, while the OpenGL space uses the physical coordinates.
> What a mess.

Missed that.

Mirarkitty, 2019-05-16:
I would have assumed https://github.com/prusa3d/PrusaSlicer/issues/2247 was a return of this but it seems intentional now

wavexx, 2019-05-16:
It might be intentional, but it's a waste of space in my eyes. I want the plate to have as much as screen real-estate as possible, I don't really have problems hitting the buttons. 

This is what I do currently:

````
diff --git a/src/slic3r/GUI/GLCanvas3D.cpp b/src/slic3r/GUI/GLCanvas3D.cpp
index bb2cb5e2d..ab1c42861 100644
--- a/src/slic3r/GUI/GLCanvas3D.cpp
+++ b/src/slic3r/GUI/GLCanvas3D.cpp
@@ -3416,7 +3416,7 @@ bool GLCanvas3D::_init_toolbar()
     }

 #if ENABLE_SVG_ICONS
-    m_toolbar.set_icons_size(40);
+    m_toolbar.set_icons_size(32);
 #endif // ENABLE_SVG_ICONS
 //    m_toolbar.set_layout_type(GLToolbar::Layout::Vertical);
     m_toolbar.set_layout_type(GLToolbar::Layout::Horizontal);
diff --git a/src/slic3r/GUI/Gizmos/GLGizmosManager.cpp b/src/slic3r/GUI/Gizmos/GLGizmosManag
index 478bd1fa7..6907a7b9f 100644
--- a/src/slic3r/GUI/Gizmos/GLGizmosManager.cpp
+++ b/src/slic3r/GUI/Gizmos/GLGizmosManager.cpp
@@ -13,7 +13,7 @@ namespace Slic3r {
 namespace GUI {

 #if ENABLE_SVG_ICONS
-    const float GLGizmosManager::Default_Icons_Size = 64;
+    const float GLGizmosManager::Default_Icons_Size = 32;
 #endif // ENABLE_SVG_ICONS

 GLGizmosManager::GLGizmosManager()
diff --git a/src/slic3r/GUI/Plater.cpp b/src/slic3r/GUI/Plater.cpp
index c63ae73f2..59511b236 100644
--- a/src/slic3r/GUI/Plater.cpp
+++ b/src/slic3r/GUI/Plater.cpp
@@ -3110,6 +3110,7 @@ void Plater::priv::init_view_toolbar()
     view_toolbar.set_layout_orientation(GLToolbar::Layout::Bottom);
     view_toolbar.set_border(5.0f);
     view_toolbar.set_gap_size(1.0f);
+    view_toolbar.set_icons_size(32);

     GLToolbarItem::Data item;
````

The buttons are still larger than a regular GTK toolbar in both low and hi-dpi modes.

Mirarkitty, 2019-05-16:
I would agree with you that they are quite a bit large now.

vojtechkral, 2019-05-17:
@wavexx @Mirarkitty what's your default GTK font size? I think this is simply because our windows devs use smaller fonts...

wavexx, 2019-05-17:
On Fri, May 17 2019, Vojtech Kral wrote:
> @wavexx @Mirarkitty what's your default GTK font size? I think this is
> simply because our windows devs use smaller fonts...

The default GTK font I'm using is "Roboto 12pt" at either 140dpi or
95dpi depending on the screen.

The resulting physical font size looks equivalent to a default windows
10 installation on the same laptop, so I don't think my font size is
that much "larger" than usual.

The legend text in the slicing preview matches the system's size, so it
looks the scaling is correct. Still, the side buttons are 4x the height
of a line of text.

tedder, 2019-05-18:
related: #2247, posting my screenshots here as well.

[image]
[image]

Mirarkitty, 2019-05-20:
Have a comparison with google chrome button sizes.
[image]

bubnikv, 2019-08-01:
Button sizes are configurable through application preferences in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv