prusa3d/PrusaSlicer #1907: Inconsitent handling of tiny structures

rennne, 2019-03-05:
When I use " 0.15mm OPTIMAL MK3" with "detect thin wall"  very tiny walls are extruded but not slightly bigger structures. 

First a very tiny wall is taken into account:
[image]
[image]

The same object with a much bigger wall, but the wall is now ignored:

[image]
[image]

I found that this tends to happens when the two sides of the little wall are not perfectly parallel.

The two stl files:
[boxAandB.zip](https://github.com/prusa3d/Slic3r/files/2932410/boxAandB.zip)

tecnopolis-ve, 2019-03-06:
I've been struggling with the same behavior, try to disable "detect thin wall perimeter"

rennne, 2019-03-08:
When I disable "detect thin wall perimeter", it just get worse:
Both structures are just ignored.

In all cases the egde is more round then a normal 90° corner, but when no wall can be applied it at least should be sharper, like a 110° degree corner and positioned more outside-> in the direction where the missing structure should be

supermerill, 2019-03-27:
it's a known issue, i try to resolve it. See https://github.com/slic3r/Slic3r/pull/4534.

closed 2026-09-29 by prusaslicer-bot[bot]