prusa3d/PrusaSlicer #1908: Slic3r does not notify users of post processing script errors

domesticatedviking, 2019-03-05:
ALL versions, all OS's
It would be nice if Slic3r communicated when post processing scripts failed.  Ideally slic3r would display the exit condition set by the script.

bubnikv, 2019-03-06:
and if slic3r showed the log collected from the post processing script in
that case.

By the way, what return code shall be considered an error? Negative for
Linux / OSX, non-zero for Windows?
On Windows, the return code is an unsigned 32bit int
https://docs.microsoft.com/en-us/windows/desktop/api/processthreadsapi/nf-processthreadsapi-getexitcodeprocess

bubnikv, 2019-03-06:
@vojtechkral

bubnikv, 2019-03-17:
Implemented for Linux / OSX by @vojtechkral , this will be part of the beta.
Implementation for Windows is special and it is still missing.

domesticatedviking, 2019-03-18:
Excellent!  Thank you for looking at this!

bubnikv, 2020-02-04:
It is supposed to be implemented long time ago. closing.

closed 2020-02-04 by bubnikv