prusa3d/PrusaSlicer #1909: Wrong Orientation after Place on Face (upside down)

dartrax, 2019-03-06:
### Version
alpha 7

### Behavior
Place on Face Feature: When clicking a Face of the part, the part should be aligned in a way that the face is the bottom. In this case, it is the top.

In the screenshots, notice the chamfer at the part, the camfer should be aligned down.

Also, there is some odd behavior with face highlighting in certain situations (see screenshot) or faces, that are indicated, but do not react on clicking or hovering.

#### Project File (.3MF) where problem occurs
[image]
[image]
[image]
[image]

lukasmatena, 2019-03-06:
Could you please attach a 3MF? Thanks.

dartrax, 2019-03-06:
done. Got filtered at first attempt ;-)

dartrax, 2019-03-06:
Hi lukasmatena, thanks for looking into it.

I've just tested again, I downloaded the "part.zip" from first post, opened a new Slicer Session, and dropped the extracted part.3mf into ist. Im still reproducing this.

I'll also attach the original from the screenshots.

It may be related to my 3D graphic system?
We are using HP Z440 Workstation.
This is the System Information:

```
Slic3r Prusa Edition
Version:   1.42.0-alpha7+win64
Build:     Slic3rPE-1.42.0-alpha7+win64-full-201903011809

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17134), 64-bit edition

OpenGL installation
GL version:   4.6.0 NVIDIA 390.77
Vendor:       NVIDIA Corporation
Renderer:     Quadro M2000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
[... 306 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

```
[testfile_large.zip](https://github.com/prusa3d/Slic3r/files/2936927/testfile_large.zip)

dartrax, 2019-03-06:
I've tried to reproduce that on another PC, this time a Dell Notebook. I'm getting a nice blue touch on one indicator ;-)

But besides from that, Place on Face is working as it should and the highlighting of all of the face indicators works as expected.

Great work with the highlighters by the way, I really like them.

[image]

bubnikv, 2019-03-17:
The blue tint indicates that there is something rotten inside the software.
@lukasmatena please test with asan

dartrax, 2019-04-02:
Just for information: Still present in 1.42.0-beta1

bubnikv, 2019-04-02:
Interesting. Did you maybe apply mirroring to the model?

dartrax, 2019-04-02:
No, just importing and selecting the Place on Face Feature.
Is there anything I can do? Like any system information or a short video of the issue?

lukasmatena, 2019-08-26:
@dartrax Could you please retest this with PrusaSlicer2.1.0-beta2? I am still unable to reproduce it but the problem could have been solved by some of the recent changes.

dartrax, 2019-08-26:
I‘ll do. It was still present in 2.1 alpha, but I‘ll test with the then-latest version the next days. Thanks for looking into it!

lukasmatena, 2019-08-26:
> Thanks for looking into it!

Not much to thank for, I looked into in but didn't find anything. I was just hoping this has been flushed along with something else. But if you still had issues in 2.1.0-alpha1, it will probably not be the case.

dartrax, 2019-08-28:
I'm still having this issue with 2.1.0-beta2. I updated the first post with a video.

As for the blue tint on the indicator with my Dell Notebook, this was solved by updating Windows and OpenGL drivers. I've updated my comment above: https://github.com/prusa3d/PrusaSlicer/issues/1909#issuecomment-470152720

bubnikv, 2019-08-28:
> I'm still having this issue with 2.1.0-beta2. I updated the first post with a video.

It looks like a picking problem. There is an issue with some NVidia cards, that do not respect disabled multi sampling, making the picking by false color unreliable. We have one computer in house, which suffers this issue, we will test tomorrow.

bubnikv, 2019-08-30:
We have tested it on that particular computer with NVidia card that used to have the picking problem before we fixed it. The "place on face" face selection works correctly. I am not sure what else could we do. Interestingly enough, you seem to be the only one reporting the issue, most likely it is caused by wonky graphics card driver.

rtyr, 2020-04-17:
@dartrax Do you still have this issue with 2.2.0?

dartrax, 2020-05-18:
@rtyr, thanks for coming back to it. Unfortunately, since we are in home office, i have only remote access to the machine, and I can't reproduce it even with 2.1.0-beta2 right now. I think the reason is: the OpenGL drivers are not the same, it shows this instead of a NVIDIA vendor:

```
OpenGL installation
GL version: 3.0 Mesa 12.0.0-rc2
Vendor: VMware, Inc.
Renderer: Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits)
GLSL version: 1.30
Installed extensions: [...]
```

dartrax, 2020-06-09:
@rtyr, yes, i do still have this issue with 2.2.0
I'm having direct access to the machine once a week now.
I've attached a video.

[video_2.2.0.zip](https://github.com/prusa3d/PrusaSlicer/files/4753487/video_2.2.0.zip)

rtyr, 2020-06-09:
Thank you for testing.

lukasmatena, 2020-06-09:
I wonder if this mysterious issue still appears in current master. The highlighted faces are no longer rendered in immediate mode as they were before. If you @dartrax had nothing else to do, you can try it out. Someone has recently shared a Windows build [here](https://github.com/Rob4226/hello-world/issues/1).

I'm sorry to bother with more and more builds to test, but like was mentioned already, you are the only one suffering from this issue that we know of. The problem is quite likely not on PrusaSlicer's side and I don't really know what to do.

dartrax, 2020-06-24:
@lukasmatena, I have tried with the build you've linked and indeed, I was not able to reproduce this issue on that machine!
So it seems to be fixed by that build. :)

closed 2020-06-24 by dartrax

lukasmatena, 2020-06-25:
@dartrax That means that OpenGL legacy immediate mode (or its implementation in your driver) ws the culprit. Thanks for letting us know, that's good news.