prusa3d/PrusaSlicer #1910: bed image not appear

davarr76, 2019-03-06:
### Version
Slic3r Prusa Edition 1.42.0-alpha7

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
original prusa i3 MK3S

### Behavior
Hi,
I have just installed the new beta version of Slic3R.
I have changed the form plate, using the .stl file in the repertory "resources/models", but I don't know how to import the high resolution SVG texture, as explained in the new features of this version.

Thanks for your help !
Regards,
David

Mike4U, 2019-03-09:
I would also like to know.

thanks

beeb, 2019-05-03:
My bed model disappeared also. I just have the rectangular grid but nothing else. I don't know how to add it back, can anyone help?

bubnikv, 2019-05-04:
they should show up automatically for Prusa3D system profiles. Does it not work for you?

davarr76, 2019-05-04:
No it doesn't work !

I have used the wizard configuration, choosing i3 MK3S, but nothing occurs ....

bubnikv, 2019-05-04:
which version did you try last? Did you extract the slic3r into a clean directory? Would you please pass a screenshot?

davarr76, 2019-05-04:
I have tried the 1.42.0-alpha7 version. See attached files.
On my computer there is also the official version installed : 1.41.3

[image]
[image]
[image]

bubnikv, 2019-05-04:
The latest one is https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta2

davarr76, 2019-05-04:
Yes, it works !!!

Thanks for your help. I didn't noticed that it had moved from alpha7 to beta2 ;) ...

Regards,
David
[image]

closed 2019-05-04 by davarr76