prusa3d/PrusaSlicer #1911: Complete individual objects: "checker" bug

supermerill, 2019-03-07:
### Version
1.42.0-alpha7

### Operating system type + version
win64

### Behavior

when i check "Complete individual objects", a error message warn me about object too near.
I then click on "arrange"(or move myself) to pull them apart (great work on that, it's much better than before)

But the error box comes back saying it's not far apart enough. I check that by modifying the source and exporting svg, it seems that the checker (Print.cpp :: std::string Print::validate() const) doesn't move the copies at the right place.

I corrected that in supermerill@86fe7e333f29aac4bbd8b575ea4508305d37c54a by grabbing the chunk of code that do the same job in ModelArrange.cpp. It works better, but there are some code duplication and i didn't test it with all combinations of move - copies - rotation - scale.

#### Project File (.3MF) where problem occurs
will upload later, i swear

bubnikv, 2019-04-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/5ec507e9eabf0d3d29821c9007a614150b025be9

@supermerill thanks for the hint. Would you please verify the solution?

bubnikv, 2019-05-10:
Hopefully my rework of @supermerill 's fix is working as @supermerill intended.
Closing.

closed 2019-05-10 by bubnikv