prusa3d/PrusaSlicer #1912: [Feature request] add suport for plater (automatic nesting software)

Spacemarine2018, 2019-03-07:
The integrated auto-arrange feature in Slic3r PE is very limited, as it can not place objects inside one another. A much better solution would be to integrate the open-source nesting software Plater. Plater is a 3D-printer parts placer and plate generator.

There is a nice demo-video in the following link:
https://github.com/Rhoban/Plater

bubnikv, 2019-03-08:
Thanks for the hint. @tamasmeszaros is currently fully occupied with the SL1 support, but I am sure he will look into the project you referenced.

tamasmeszaros, 2019-03-08:
Hello! Plater is a great project and it is a reasonable request. On the other hand, our implementation enables us to fine tune the shape of the nested pile of objects and there are other very delicate requirements for the arrangement result which is hard to satisfy with an outsourced library. The extension of the Slic3r nesting module with part-in-part placement is on the roadmap as well as the concave geometry support.

closed 2020-03-16 by tamasmeszaros