prusa3d/PrusaSlicer #1913: [Feature Request] Some ideas for the preview

th33xitus, 2019-03-07:
### Version
1.42.0-alpha7+win64

### Idea 1:
At the moment you need to click the upper or lower semicircle to "toggle" scrolling through the line-movement commands:
[image]

Depending on where you move your cursor to after "toggling" the scrolling, scrolling either works or doesn't work:
[image]

The problem now is simple. When i scroll "upwards" i always have to make sure that my cursor is on top of the semicircle (or below if i select the lower semicircle) otherwise scrolling will stop as soon as this case occurs:
[image]
The green arrow is indicating the scroll direction, i want to increase the movement count, but scrolling is disabling itself as soon as the cursor is at the indicated position or lower.

What about to make it that it doesn't matter where you move your cursor to while you selected the semicircle and therefor toggling the scrolling function?

### Idea 2:
Radiobutton or dropdown menu to switch between "layer-view" or "line-movement-view" so you either scroll through each single movement or scroll through complete layer. 
For example like this:
[image]
So depending on which radiobutton is active you either scroll through single lines or single layer.

### Idea 3:
Include a stilised nozzle model in the preview so you can see where the nozzle is at that moment. Something like i simple upside down cone would be sufficient as "nozzle position indicator" imho.

### Idea 4:
Include an animation feature. Make a play/pause button and a speed slider to set animation speed. When you click it the program then will "scroll" through each line movement by itself starting from the last nozzle position you selected. This would be some kind of simulating all of the movements of a whole print.

YuSanka, 2019-03-11:
Thanks for your report. This misunderstood behavior was fixed. 
Current behavior is to move selected thumb only. If no one thumb is selected, then will be selected the thumb nearest to the current mouse position.

closed 2019-03-14 by bubnikv

SteWhite, 2020-04-15:
about **idea 2** - why won't it be implemented? the simulation line by line  is very useful to solve some printing problems. I have it in Simplify3D and I miss it awfully.