prusa3d/PrusaSlicer #1914: [Feature Request] Remember last profile compatibility state

th33xitus, 2019-03-07:
### Version
1.42.0-alpha7+win64

### Idea:
At the moment, the profile compatibility state is discarded when closing and reopening the program.
When i switch it so that both, compatible and incompatible printing profiles are shown (red flag), this status won't be saved and needs to be set every time i open the program.

Is there any specific reason why this state isn't saved upon closing?

bubnikv, 2019-03-08:
We cannot reproduce it. Would you please give us more detail? Step by step workflow, files, screen shots.

th33xitus, 2019-03-08:
### Operating System:
Windows 7 Ultimate 64bit

### Configuration -> Preferences:
[image]

### Step 1:
Open program, check profile compatibility state:
[image]

### Step 2:
Click on green flag to switch the mode:
[image]

### Step 3:
Close program

### Step 4:
Reopen program, check compatibility state.
For me, same picture is to be seen as in **step 1**

I also tried if starting the program as admin makes any difference: it doesn't.

What files do you need?

bubnikv, 2019-03-08:
Thanks for the details.

The behavior is as intended. The red state of the flag is temporary on
purpose, it is there just to help you to debug the dependencies, it is not
intended to be a permanent workaround to the configuration issues.

th33xitus, 2019-03-08:
Hm okay, somewhat sad.
Personally i don't see it as some kind of workaround with the red flag enabled, i rather like to see all my custom created profiles on first glance wether they are compatible to my current printer/print settings or not. But okay, i then have just to re-enable it every time i start the program.

bubnikv, 2019-03-08:
I am sorry I made you sad.

Closing.

closed 2019-03-08 by bubnikv