prusa3d/PrusaSlicer #1915: [1.42.0-alpha7] SLA struts fail on mirrored object

Maeyanie, 2019-03-08:
### Version
version_1.42.0-alpha7

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_
When working on an object which has been mirrored within Slic3r, SLA struts do not properly generate.

* _Steps needed to reproduce the problem_
1. Add an object.
2. Right-click the object, Mirror, Mirror along [XYZ] axis (which axis doesn't matter)
3. Auto-generate support struts. Note that the points generate in sensible spots for the mirroring.
4. Click "Slice now."

* _Expected Results_
Struts generate properly according to the placement of strut support points.

* _Actual Results_
[image]

_Is this a new feature request?_
No.

Maeyanie, 2019-03-17:
Problem is still present in -beta, updated versions in the post.

bubnikv, 2019-03-17:
We know, we did not have time to fix it yet. It is in the planning for the
next beta.

Maeyanie, 2019-03-17:
It's definitely not a high-priority thing, I was just making sure the report was still relevant by testing it against the new version.

Is there a "known problems" list anywhere, so I can avoid reporting them in the future?

bubnikv, 2019-04-02:
> Is there a "known problems" list anywhere, so I can avoid re-reporting them in the future?

That's basically the github issue tracker.

This issue has been fixed on the master, the fix will be part of the next beta. Closing.

closed 2019-04-02 by bubnikv