prusa3d/PrusaSlicer #1917: UI scaling is broken (probably after HiDPI changes)

volca02, 2019-03-08:
### Version
Slic3rPE-1.42.0-alpha7+linux64-full-201903011813

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Linux - ArchLinux. 
My display is fairly low-dpi (81x80 dpi)

```
xdpyinfo | grep -B 2 resolution
screen #0:
  dimensions:    2560x1080 pixels (802x338 millimeters)
  resolution:    81x81 dots per inch
```

### Behavior
[image]
[image]

* The icons are barely visible, as are some of the texts on screen. Everything is cramped up together. Some UI elements are not usable because of their size. 

My xresources is set properly for 81 DPI:
```
# xrdb -q                  
Xft.dpi:	81
...
```

bubnikv, 2019-03-09:
it is cramped because now the layout is being derived from your default
font size, namely the width of the m letter. You are the first person
reporting having an issue with it. With the information on hand about the
scaling algorithm, will you be able to solve your issue yourself, or do you
we a need for a change in the code? We can implement some minimum sizes of
the controls, minimum allowable scaling, default font selection, but I
personally would rather do no change at all if possible.

volca02, 2019-03-09:
Thanks for the reply. The thing is most of the texts are scaled properly, what's an issue for me seems to be non-text elements - buttons with icons are way smaller than they should be in proportions to the text elements.

I will try poking through the code myself.

volca02, 2019-03-09:
After hardcoding 10 (or commenting out) the em unit detection/setting in GUI_App.cpp (set_em_unit call), I'm getting a more consistent look, although the buttons with mode selection are still misbehaving.

[image]

The 'm' character size detection `GetTextExtent("m")` returns `5x12`, btw.

I'll dig a bit more later.

volca02, 2019-03-09:
The `PrusaModeButton` text overlapping with icon is caused by an unexpected forced border the button has - the calculation in `PrusaModeSizer` does not expect any button border and so it scales the button to a smaller size than what's actually needed. This might be related to GTK theming, I will try to dig further.

Additionally the calculation may also screw up the button height in cases when the text is smaller than the icon - `std::max` over `button.second.GetHeight()` (the icon size) and the text height would probably be a more robust solution.

I've made a small tweak to the initialization code to allow env. variable usage to override the calculation in order to fix the scaling on systems with unexpected display DPIs.

```
diff --git a/src/slic3r/GUI/MainFrame.cpp b/src/slic3r/GUI/MainFrame.cpp
index cddf081eb..3c4d0b793 100644
--- a/src/slic3r/GUI/MainFrame.cpp
+++ b/src/slic3r/GUI/MainFrame.cpp
@@ -55,8 +55,23 @@ wxFrame(NULL, wxID_ANY, SLIC3R_BUILD, wxDefaultPosition, wxDefaultSize, wxDEFAUL
 
 
     // initialize default width_unit according to the width of the one symbol ("x") of the current system font
-    const wxSize size = GetTextExtent("m");
-    wxGetApp().set_em_unit(size.x-1);
+
+    size_t em_unit_scaling = 100; // ~percentage
+
+    const char *uiScaling = boost::nowide::getenv("SLIC3R_UISCALE");
+
+    if (uiScaling) {
+        em_unit_scaling = atoi(uiScaling);
+
+        // invalid values
+        if (em_unit_scaling < 1)
+            em_unit_scaling = 100;
+    } else {
+        const wxSize size = GetTextExtent("m");
+        em_unit_scaling = (size.x - 1) * 10; // 10 is normal, we're expecting percentage here
+    }
+
+    wxGetApp().set_em_unit(em_unit_scaling / 10);
```

volca02, 2019-03-09:
GTK+ buttons having a border is a reported bug:

https://trac.wxwidgets.org/ticket/17683

volca02, 2019-03-09:
Thinking about it more - downscaling the UI is probably not an expected behavior of the change. Maybe enforcing minimum `em_unit = 10` would fix this for systems like mine without any complex effort throughout the code. For example the Platter Sidebar is shrunk too much on my display as well, meaning some UI elements are not even accessible.

bubnikv, 2019-03-11:
We will likely implement a minimum em_unit.

volca02, 2019-03-11:
Okay, thanks!

volca02, 2019-03-15:
Small update: I dug through my `.gtkrc-2.0` files and found out I had a monospace font of size 8 set as my screen font:

```
style "font"
{
font_name = "Envy Code R 8"
}
```

Commenting that out fixed the visuals of Slic3r for the price of ugly default gtk font (I'm used to that monospace font). I figured you might want to know this piece of info.

vojtechkral, 2019-03-15:
@volca02 Thanks, but I don't there's very much we can do about that. From the screenshots it seems the letter `m` is smaller than the other letters for some reason. In fact, it looks like a rectangular box to me rather than a letter `m` :) I would blame the problme on the particular font. Have you tried a different monospaced font?

volca02, 2019-03-15:
Just tried different monospaced fonts, and it is not very consistent - many (not all) monospace fonts with lower pitch will cause the UI to be compressed, but some seem fine (not sure however if all are bitmap based).

I guess this is caused by the horizontal pixel count on bitmap based fonts with smaller pitch being low, but can't be sure without digging too deep.

To be honest I find the problem as it is now marginal in importance - I'd still be for the limitation in the scaling down (em_unit to be at least 10) to avoid the icons being too small on any screen/font combination, but that's it.

bubnikv, 2019-03-24:
We set the em unit to a minimum of 10 points on the master now. This shall solve your issue.
Closing.

closed 2019-03-24 by bubnikv