prusa3d/PrusaSlicer #1918: [1.42.0-alpha7] Filename creation broken

th33xitus, 2019-03-08:
### Version
1.42.0-alpha7+win64

### Operating system type + version
Windows 7 Ultimate 64bit

### Behavior
Filename creation either via "Export G-code..." or "Send to printer" is broken when using configuration variables in the output filename like `filament_diameter` , `extrusion_multiplier` or `layer_height`.
Probably due to the `.` printed out by using those variables.

My common output filename format is:
`[input_filename_base]_FD[filament_diameter]_EM[extrusion_multiplier]_LH[layer_height]`
Which usually resulted in filenames like:
`Filename_FD1.75_EM0.9_LH0.1.gcode`

Now, when i export the G-code to my local hard drive i will get a `*.1` file due to the `.1`-ending of my variables. The program misses adding the `.gcode` at the end.
When i try to send a file directly to the printer it results in an error message.

[image]

[image]

[image]

bubnikv, 2019-03-11:
The order of the template processing is:
First replace the templates
Second add the default .gcode extension if there is none.

The problem here is, that you did not provide the .gcode file extension as
part of your template.

Now the question is, whether just flipping the order of the steps above is
a good idea, as it would break if you used a template to generate the file
extension by a template containing a dot in the template definition. The
correct solution likely would be to modify the post processor to
differentiate between the dot outside of the template from the dot inside
the template, and then to consider the file extension based on that
information. Such a change is not a 5 minutes task though.

th33xitus, 2019-03-11:
It's completely fine, i just didn't think of just adding the ".gcode" at the end by myself and instead expected the software to do it on it's own. So... actually there is no need to change anything at all, just remember to keep the ".gcode" at the end. I must have deleted it by accident while modifying the output name.

Thats why i closed this issue before any reply came it, because i don't see it as an issue anymore.

bubnikv, 2019-03-11:
The current behavior is not good, it is quite confusing, we should change
it somehow.

The software would add .gcode if it did not find a dot in the string
generated by the template processor.

th33xitus, 2019-03-11:
Okay i just undo my edit of the first post and recreated the proper topic description.

I agree with changing it, would make things easier and you still would get a proper gcode-file as output if people miss the ".gcode" at the end.

reloxx13, 2019-05-15:
(mixing with #2229)

"send g-code" is only for octoprint right? 
in this case:
does octoprint accept something else than .gcode? nope. 
does this "feature" make sense now? nope.

what does the btn say "send **g-code**" ? that it sends a **gcode** file.

i open a STL file, why would it even have a "custom" fileextension in the stl filename?

jimmy00784, 2019-05-15:
Here is the issue 
Slic3r/src/libslic3r/PrintBase.cpp:61
```
// Generate an output file name based on the format template, default extension, and template parameters
// (timestamps, object placeholders derived from the model, current placeholder prameters, print statistics - config_override)
std::string PrintBase::output_filename(const std::string &format, const std::string &default_ext, const DynamicConfig *config_override) const
{
    DynamicConfig cfg;
    if (config_override != nullptr)
    	cfg = *config_override;
    PlaceholderParser::update_timestamp(cfg);
    this->update_object_placeholders(cfg);
    try {
        boost::filesystem::path filename = this->placeholder_parser().process(format, 0, &cfg);
        if (filename.extension().empty())
        	filename = boost::filesystem::change_extension(filename, default_ext);
        return filename.string();
    } catch (std::runtime_error &err) {
        throw std::runtime_error(L("Failed processing of the output_filename_format template.") + "\n" + err.what());
    }
}
```
The logic check `if(filename.extension().empty()` is inadequate as any file name with a period in the middle of the name will pass this check. 
See its documentation [here](https://www.boost.org/doc/libs/1_45_0/libs/filesystem/v3/doc/reference.html#path-extension)
At least check if the extension matches a white list.
This needs to be re-opened

bubnikv, 2019-05-16:
Would you please provide the white list?

jimmy00784, 2019-05-16:
@bubnikv Currently how many output formats does PrusaSlicer support besides .gcode?

bubnikv, 2019-05-16:
I did the following changes in https://github.com/prusa3d/PrusaSlicer/commit/df19314645541611f1e51381fd48469122897bd6

I changed the default of output_filename_format to "[input_filename_base].gcode" for FDM, and "[input_filename_base].sl1" for the SLA technology. This gives the user a clear clue, that the extension should better be provided in the template field. Now the user gets the output_filename_format filled in with the correct extension even if he or she adds some dots to the default template.

Also from now if the output_filename_format is empty, the output name is generated, and the correct extension (gcode or sl1) is attached.

Closing.

closed 2019-05-16 by bubnikv

bubnikv, 2019-05-16:
I know of one firmware, that requires the gcode to be renamed, but for the
life of me I cannot remember.
Then there are G-code post processors, and the Smoothieware uses a binary
G-code with a different extension.
There may be more exceptions out in the wild.

jimmy00784, 2019-05-16:
@bubnikv I still like you approach better. You've covered the bases for extensions that are standard and supported out of the box. For those that require different extensions than gcode, the configuration settings for output file name would have to be altered anyways, so I would supposed that functionality wouldn't break. Looking forward to testing this in the next release.

jimmy00784, 2019-05-16:
@bubnikv This works as long as the "Print Settings" > "Output options" > "Output filename format" is empty. I have successfully tested this on a local build from master.
The default configuration for new "Print Settings" will have to have "Output filename format" as empty as well or this won't be a complete fix by itself.