prusa3d/PrusaSlicer #1919: [1.42.0-alpha7] Unexpected solid infill on vertical walls

BirkBinnard, 2019-03-08:
### Version
1.42.0-alpha7

### Operating system type + version
WIn10/64 current version

### Behavior
Unexpected solid infill is produced in STL with vertical shell

[image]

The STL file for this part:  [https://www.thingiverse.com/thing:3478042](url)

bubnikv, 2019-03-11:
I don't see it. A 3MF would be helpful as it contains all the settings you chose for your print.

BirkBinnard, 2019-03-11:
[image]

It's very interesting that the 3MF file behaves differently - only one loop has solid infill instead of 2 when slicing the STL file.

It's also interesting how much smaller the 3MF file is - should we all switch from STL to 3MF now? 

Here is the link to the 3MF file: https://drive.google.com/file/d/1OAFlu2uQV3Q0lYNlftQpPof2xs5EbdjS/view?usp=sharing

Is there any chance this software can be tweaked to allow attaching 3MF & STL??

closed 2019-03-11 by BirkBinnard