prusa3d/PrusaSlicer #1920: Slic3r Prusa Edition 1.42.0-alpha7 layer gaps

GeekDad63, 2019-03-09:
### Version
_Version of Slic3r Prusa Edition used goes here_
1.42.0-alpha7
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Mac OS 10.13.6

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa i3 MK3 Firmware V3.5.1

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
Slicing object produces gaps in the slices as seen in the screen shot. I have also sent a screenshot of the same model in 1.41.3+ which is the lower screenshot.
[image]
[image]

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-03-11:
Please attach your 3MF. Thanks.

GeekDad63, 2019-03-11:
[botmount-top-52mmlens-screw_in.zip](https://github.com/prusa3d/Slic3r/files/2952481/botmount-top-52mmlens-screw_in.zip)

GeekDad63, 2019-03-11:
[botmount-top-52mmlens-screw_in.3mf.zip](https://github.com/prusa3d/Slic3r/files/2952514/botmount-top-52mmlens-screw_in.3mf.zip)

bubnikv, 2019-03-11:
This seems to be a single line regression, fixed with c4829338453691371fc089d0aaf2cb9271417415

The fix will be part of the upcoming beta. Closing.

closed 2019-03-11 by bubnikv