prusa3d/PrusaSlicer #1921: flying supports

OUPS65, 2019-03-09:
Hope will be helpfull
Supports generated by current 1.41.3 version as well as the  1.42.0 alpha 7  are not usable for my model : they generate some supports  flying «on the air» and  they do not necessarily supports something.  Some one are also useless.
The experimental tree supports generated by CURA are lovely more sophisticated supports wich are doing a perfect job.
Parameters for both programs : 
STL File « fixed.obj » corrected with the internal Prusa STL repair process
 005 ultra details, HIPS, 0.25 nozzel 
Filament HIPS : not soluble material
Supports : Z « 0,2 detachable » and « XY separation between an object and its support » : set to 1mm

with current 1.41.3 version 
[image]

with 1.42.0 alpha 7
[image]

[SUPPORTS.zip](https://github.com/prusa3d/Slic3r/files/2948345/SUPPORTS.zip)

yarus23, 2019-03-19:
yet another test

[test2.zip](https://github.com/prusa3d/Slic3r/files/2985649/test2.zip)
[image]

F0x06, 2019-05-22:
[image]

[image]

Same issue with https://www.prusaprinters.org/prints/1107-prusa-mk3-heatbed-cable-support-easy-clip-on
Supports: On build-plate only

closed 2026-09-29 by prusaslicer-bot[bot]