prusa3d/PrusaSlicer #1922: Estimated print time command line

DreiDe, 2019-03-09:
Is it possible to get the estimated print time as well as the estimated weight from the command line interface as it is possible in the GUI? I could't find any option for doing that. If it's not possible yet I would suggest to add it as a feature request.

bubnikv, 2019-03-10:
you should be able to parse it from the exported G-code. Is it not
sufficient?

DreiDe, 2019-03-10:
Oh, sorry I didn't notice that there is an `filament used` parameter inside the Gcode. And regarding to this I also found this comment: https://github.com/prusa3d/Slic3r/issues/922#issuecomment-446642499 mentioning also a `print_time` parameter.

bubnikv, 2019-03-11:
> the Gcode. And regarding to this I also found this comment: #922 (comment)

#922 only regards the parameters to be used inside the file name template.
That may help you as well though.

rtyr, 2021-01-26:
Seems answered. Closing.

closed 2021-01-26 by rtyr