prusa3d/PrusaSlicer #1923: [Feature Request] Addional Extrusion Width Option

dot-bob, 2019-03-09:
Add an option to the Advanced settings under Extrusion width to have the extrusion width percentage follow a percentage of "Layer height" (like it is now) or follow the Printer settings "Nozzle diameter".

I have many printers with different nozzle diameters and all that is different between print settings is the advanced tab extrusion width.  This would allow use of the same print settings for multiple sized nozzles.

guestisp, 2019-03-15:
I was opening the same exact issue right now. I need the same thing. In example, I need to set a default extrusion width (so that any other extrusion width are related to) to, in example, 99% of nozzle diameter or 101% of nozzle diameter and not on layer height.

The current behaviour, when using percentage, is to evaluate against the layer height.

towe96, 2019-12-09:
Still very relevant.

I have no idea why extrusion width should ever be relative to layer height rather than nozzle diameter.

Photogad, 2021-02-12:
+1

It would be a huge quality of life improvement to have 1 print profile for multiple nozzle sizes instead of having to make 8 different profiles like me.  Change one setting and you gotta do it 8 times, it's such a huge annoyance.

Neee widths to be % of nozzle diameter, and ALSO first layer height able to be set as a % of nozzle diameter  (best first layer height is almost always 50% of nozzle diameter)

Schild0r, 2022-01-31:
any updates to this topic @bubnikv? Is this something that is being worked on?
The "breaks compatibility with old profiles that use the % of layer height" argument could be resolved by having an option that toggles between the "% of layer height" and the "% of nozzle dia" logic with the former being the default?

If this will not be worked on that is a huge plus for superslicer which I for the longest time refused to adopt but with multiple different nozzle sizes in one machine this might just be the the way to go then...

bubnikv, 2022-02-01:
We are going to look into this topic for 2.5 release, at least we will do an analysis of how to handle the incompatibilities. Let's see what we will come up with.

Schild0r, 2022-02-01:
> We are going to look into this topic for 2.5 release, at least we will do an analysis of how to handle the incompatibilities. Let's see what we will come up with.

Thank you very much. Appreciate the effort taken :)

Bumblebiber, 2022-07-29:
I agree! Went back to Prusaslicer yesterday after a few months of using Superslicer and I was so surprised about how line width percentages are computed! Makes absolutely no sense.

Gonna go back to SS after the merge to PS 2.5

3D1-Production, 2023-02-16:
> We are going to look into this topic for 2.5 release, at least we will do an analysis of how to handle the incompatibilities. Let's see what we will come up with.

Has there been any more on this?

Schild0r, 2023-10-22:
Hey, just wanted to ask if this four and a half year old FR is actually still on the list or if this was scrapped now almost everyone uses arachne
As far as I am aware this is implemented in forks of Prusaslicer so it is doable and for the backwards compatibility issue I proposed a solution at the beginning of last year, which I have not heard a counter argument against.

Prusaslicer really is pushing features that I had never imagined a free software could sustain and that I'd maybe expect for a professional software like Materialise Magics, however on this background, I find it kinda ridiculous that this is still not addressed.

filamonkey, 2024-09-06:
I would rather prefer to have the extrusion width percentage linked to the nozzle diameter, as you can change nozzles and not manually have to recalculate all your widths. As it is now, if you have it linked to layer height, it will sometimes lead to terrible over-extrusion: Nozzle 0.4 mm Layer Height: 0.3 Layer width: 133% = 0.399 / Change Layer Height:0.1 Layer width = 0.133 which will overextrude to 0.4 width. I don't really see the usefulness for me right now. If you don't want to break previous profiles, just include an option to set to either.

lukasmatena, 2026-09-01:
Changed in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena