prusa3d/PrusaSlicer #1925: [Feature request]Add variable filament_type

vlcty, 2019-03-10:
### Version
1.41.3+

### Operating system type + version
MacOS 10.14.3

### Behavior
There are different variables or placeholders which can be used in custom G-code sections. I found a list here: http://mauk.cc/mediawiki/index.php/Slic3r_placeholders

There are already filament specific placeholders (like colour) there, but there is no easy way to get the filament type like PLA, ABS, PETG, etc. The info is already present in slic3r but not accessable as a variable. I think if the colour is accesable by a variable then type should also be available.
My specific use case is that I only want to apply custom G-code templates when printing with PLA.

Expected behaviour: The currently used filament type is accessable via `filament_type`.

bubnikv, 2019-03-11:
It works already, you just need to be aware that the filament specific
fields are vector variables due to the multi-extruder support. Thus

; Filament Type: {filament_type[0]}

will add the filament type comment

vlcty, 2019-03-12:
Ah okay. The variable `filament_type` is already present! Thougt it was not. Thanks!

closed 2019-03-12 by vlcty