prusa3d/PrusaSlicer #1926: Fan is mostly off

yarus23, 2019-03-10:
### Version
1.42-alpha 7+

### Operating system type + version
Mac OS 10.14.3

### 3D printer brand / version + firmware version (if known)

Ender 3 pro / Marlin 2 git

### Behavior

I have noticed that fans is 90% off although in a settings there is Auto fan is on and min is 35%
G code looks like 
`M106 S255
G1 F1800
G1 X128.988 Y94.396 E0.00340
M107
G1 F2700
G1 X129.058 Y94.529 E0.00509
M106 S255
G1 F1800
G1 X129.752 Y95.893 E0.06397
M107
G1 F2700
G1 X129.849 Y96.153 E0.00939
M106 S255
G1 F1800
G1 X130.123 Y96.896 E0.03311
M107
G1 F2700
G1 X130.400 Y97.646 E0.02706
G1 X130.641 Y99.537 E0.06452
G1 X130.187 Y101.117 E0.05566
G1 X130.065 Y101.281 E0.00691
G1 X129.874 Y101.159 E0.00766
G1 X128.817 Y103.074 E0.07406
G1 X127.925 Y103.833 E0.03962
G1 X127.150 Y104.324 E0.03107
G1 X125.825 Y105.359 E0.05689
G1 X124.864 Y105.954 E0.03825
G1 X123.414 Y106.196 E0.04977
G1 X121.030 Y105.637 E0.08291
G1 X118.842 Y104.676 E0.08088
G1 X117.851 Y103.722 E0.04653
G1 X116.842 Y103.020 E0.04164
G1 X115.615 Y104.251 E0.05882
G1 X115.266 Y104.663 E0.01829
G1 X115.243 Y104.223 E0.01493
G1 X115.447 Y103.771 E0.01678
G1 X115.887 Y103.016 E0.02959`

Why M107 is used at all if min fan is 35% in a settings?

bubnikv, 2019-03-17:
Please attach a 3MF where we could reproduce your issue.

ManuelMcLure, 2019-03-17:
Did you also set "Keep fan always on"? I found that if that is not set Slic3r will still turn off the fan if the layer time is long enough. The description below the "Keep fan always on" and "Enable auto cooling" check boxes describes that behavior.

yarus23, 2019-03-19:
I have not enabled Keep fan always on, automatic mode should not disable fan at all according its settings (min 35%(

[test.zip](https://github.com/prusa3d/Slic3r/files/2985555/test.zip)

ManuelMcLure, 2019-03-19:
That's not what the description shows if you have it turned off. If I turn it off the last line of the description is "During the other layers, fan will be turned off" even though I have Fan Speed Min set to 75%. If I enable "Keep fan always on" the last line of the description says "During the other layers, fan will always run at 75% except for the first layer".
So you can interpret "Fan speed Min" as "the minimum speed of the fan if it is on at all". The reason you might want a minimum fan speed even though you're sometimes turning off the fan is that some PWM fans don't work well below a certain threshold speed.

bubnikv, 2019-04-13:
@yarus23 is it clear? Can we close this issue?

closed 2019-04-13 by yarus23