prusa3d/PrusaSlicer #1928: Don't print support smaller than

espr14, 2019-03-10:
### Version
_Version of Slic3r Prusa Edition used goes here_
1.42.0 alpha 7
_Use `About->About Slic3r` for release versions_
Win64

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 7

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
i3 MK3

### Behavior
* _Describe the problem_
Even the smallest support is printed. It is laborious to remove all these small supports. Moreover, not printing them won't affect quality of result.
* _Steps needed to reproduce the problem_
Create object with 1 mm (or similar) overhang, enable supports and slice it.
* _Expected Results_
Small support won't be printed if you set so.
* _Actual Results_
Support is created.
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_
Yes. Please, add option to support menu: "Don't create support smaller than:". It could be in mm (length of string) or mm^2. If you set e,g, 2 mm, it won't create small supports.

espr14, 2019-03-10:
Whole support column for no help.
[image]

bubnikv, 2019-03-11:
It is not as simple as you think. The current algorithm is very local. If
your recommendation was implemented, a sharp long sloping ridge would not
be supported at all.

What we need is to extend the local algorithm to calculate the areas
needing supports into something, which is aware of the non-local properties
of the shape. And that is complex to do indeed.

espr14, 2019-03-11:
I suppose slicer does:
1. create model
2. detect overhang
3. create support interface
4. create support for the interface

If you mask all small overhangs in step 2 it won't create corresponding supports.

dzieva, 2019-03-11:
Overhangs seems to be a major problem, as I mentioned in another post, "Detect bridging perimeters" could be numerical, zero to disable and a value above zero would adjust the flow for the overhangs, and not use the bridge flow parameter.

The overhang perimeter sometimes only needs half the value of bridge_flow_ratio.

bubnikv, 2019-03-11:
>  If you mask all small overhangs in step 2 it won't create corresponding
supports.

But you will also get no supports where you expect them.

espr14, 2019-03-12:
It will remove small ones so supports will be generated for the big ones.