prusa3d/PrusaSlicer #1930: Unify support and infill patterns

espr14, 2019-03-10:
### Version
_Version of Slic3r Prusa Edition used goes here_
1.42.0 alpha 7
_Use `About->About Slic3r` for release versions_
Win64

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 7

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
i3 MK3

### Behavior
* _Describe the problem_
There are a lot of infill patterns but just few of support patterns. Still, it does very similar thing.
* _Steps needed to reproduce the problem_
Compare infill pattern menu and support pattern menu.
* _Expected Results_
List of infill and support patterns are the same.
* _Actual Results_
There are only 3 support patterns.

_Is this a new feature request?_
Yes

bubnikv, 2019-03-11:
How would it help?

espr14, 2019-03-11:
Less code (less issues) for developers, more options for users and less confusion because of different naming (rectilinear vs. rectilinear grid). Current rectilinear grid support pattern has issues but equivalent infill pattern has not. Also newly added infill patterns will be available for support immediately.

closed 2021-02-19 by bubnikv