prusa3d/PrusaSlicer #1931: [1.42.0-alpha7] After splitting a model extruder selection is ignored if you remove all but one of the split parts

riskable, 2019-03-11:
### Version
1.42.0-alpha7

### Operating system type + version
Ubuntu Linux 18.10

### 3D printer brand / version + firmware version (if known)
Prusa MK25 MMU 2.0

### Behavior
* After using the "Split to parts" feature changing the extuder on the various split parts does nothing if you remove all but one of the split parts.
* Reproduce: Split a part then remove all but one and try changing the extruder.  You can change the value but it doesn't *actually* change the extruder that will be used on the part.  The resulting gcode will have whatever extruder it had before you got down to one part.

I made a screen recording of the problem in action: https://gfycat.com/PastelSimplisticAnemoneshrimp

_Is this a new feature request?_ No.

#### Project File (.3MF) where problem occurs

[the_problem.3mf.zip](https://github.com/prusa3d/Slic3r/files/2952992/the_problem.3mf.zip)

YuSanka, 2019-03-13:
Thanks for your report. Working on it in the [SPE-860](http://dev.prusa3d.com/browse/SPE-860)

FidelCapo, 2019-03-27:
Thank you for report. This issue should be fixed in next release.
Closing.

closed 2019-03-27 by FidelCapo