prusa3d/PrusaSlicer #1932: Generic modifier object parameters propagate to primary object - no top layer on primary object

jorgeruimachado, 2019-03-11:
### Version
1.41.3+

### Operating system type + version
MACOSx, 10.14.3

### 3D printer brand / version + firmware version (if known)
Prusa MK3S firmware 3.6.0

### Behavior
hen adding a modifier object to create a wall around the primary object, with a single perimeter wall, 0% infill 
-Bottom solid layer=0
-Top solid layer=0
-perimeter=1
-infill=0

The main object will not show any top layer has solid, they will show the infill mesh.
This happen without touching on settings the primary object parameters.
See pictures:

https://ibb.co/fXYysTP
https://ibb.co/pJfTkK2
https://ibb.co/1MWWX4T
https://ibb.co/XZb4shC
https://ibb.co/RNJxykg
https://ibb.co/bBC06Lb

_Is this a new feature request?_

https://www.dropbox.com/s/d3tvtiy9gjpp84t/x-end-idler_0.2mm_PET_MK3S.3mf?dl=0

bubnikv, 2019-03-17:
I see. This is an interesting bug. Somehow the number of top/bottom layers propagate from one volume to the other.

A workaround for you is to print the cylinder as a separate object.

jorgeruimachado, 2019-03-17:
I know and that's what I've done.
This bug is on the latest slic3r 1.42.0 alpha 7

bubnikv, 2021-06-03:
This is an old one. Tested with PrusaSlicer 2.4.0-alpha1, it seem to work fine. Only in your 3mf the cylinder is marked as printable volume, not modifier. I had to switch its type to a modifier to achieve the following screenshot.

[image]

closed 2021-06-03 by bubnikv