prusa3d/PrusaSlicer #1933: Slic3r crashes when moving the layer slider and accidentally right-click in parallel

mloidl, 2019-03-11:
### Version
version_1.42.0-alpha7-99-gba3b06599

### Operating system type + version
Linux x1 5.0.0-arch1-1-ARCH #1 SMP PREEMPT Mon Mar 4 14:11:43 UTC 2019 x86_64 GNU/Linux

### 3D printer brand / version + firmware version (if known)
Issue is not printer relevant

### Behavior
When moving the Layer-Slider and in parallel right click the following assertion occurs

```
ASSERT INFO:
./src/common/wincmn.cpp(3271): assert "!wxMouseCapture::IsInCaptureStack(this)" failed in CaptureMouse(): Recapturing the mouse in the same window?

BACKTRACE:
[1] wxWindowBase::CaptureMouse()
[2] PrusaDoubleSlider::OnRightDown(wxMouseEvent&)
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[9] g_closure_invoke
[10] g_signal_emit_valist
[11] g_signal_emit
[12] gtk_propagate_event
[13] gtk_main_do_event
[14] g_main_context_dispatch
[15] g_main_loop_run
[16] gtk_main
[17] wxGUIEventLoop::DoRun()
[18] wxEventLoopBase::Run()
[19] wxAppConsoleBase::MainLoop()
[20] wxEntry(int&, wchar_t**)
[21] main
[22] __libc_start_main
[23] _start
```

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Occurs on every model.

bubnikv, 2019-03-17:
@YuSanka would you please look into it? Thanks.

bubnikv, 2019-04-13:
Does it happen with the current beta1?

mloidl, 2019-04-14:
Hmm, i've tried the current beta1 (AppImage) and it don't crash, but when i compile Slic3r myself (current master) it always gives me the exception when i right click while moving the slider.

I suspect that this is related to wxWidgets 3.0 which i use for compiling.

BeldrothTheGold, 2019-06-22:
Hello All, 

I am able to replicate this crash in a build compiled on Windows 10 from source. 

I am using Visual Studio Community 2017 (15) 64-bit and a set of dependencies built manually using the directions here: https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md 

It looks like the cmake script downloads the patched 3.1.1 wxWidgets build from the prusa3d repo.

> ExternalProject_Add(dep_wxwidgets
>     EXCLUDE_FROM_ALL 1
>     GIT_REPOSITORY "https://github.com/prusa3d/wxWidgets"
>     GIT_TAG v3.1.1-patched
> 

Interestingly, I don't see the crash when I run the release version from GitHub as mloidl commented. 

GIF showing how to replicate the crash. The crash happens when you right click while dragging in the DoubleSlider. 

[image]

This is what happens in the 2.0.0 build from GitHub. 

[image]

<details>
<summary>Assert Dialog: wxWidgets Debug Alert</summary>
[Window Title]
wxWidgets Debug Alert

[Main Instruction]
A debugging check in this application has failed.

[Content]
..\..\src\common\wincmn.cpp(3300): assert "!wxMouseCapture::IsInCaptureStack(this)" failed in wxWindowBase::CaptureMouse(): Recapturing the mouse in the same window?

[^] Hide details  [ ] Don't show this dialog again  [Stop] [Continue]

[Expanded Information]
[01] wxWindowBase::CaptureMouse              	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\wincmn.cpp:3299
[02] DoubleSlider::OnRightDown               	c:\users\scott\git\prusaslicer\src\slic3r\gui\wxextensions.cpp:2416
[03] wxAppConsoleBase::CallEventHandler      	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\appbase.cpp:670
[04] wxEvtHandler::ProcessEventIfMatchesId   	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp:1398
[05] wxEvtHandler::SearchDynamicEventTable   	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp:1868
[06] wxEvtHandler::TryHereOnly               	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp:1589
[07] wxEvtHandler::ProcessEvent              	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp:1499
[08] wxEvtHandler::SafelyProcessEvent        	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp:1617
[09] wxWindow::HandleMouseEvent              	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp:5550
[10] wxWindow::MSWHandleMessage              	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp:3064
[11] wxWindow::MSWWindowProc                 	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp:3758
[12] wxWndProc                               	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp:2847
[13] CallWindowProcW                         
[14] DispatchMessageW                        
[15] IsDialogMessageW                        
[16] wxWindow::MSWProcessMessage             	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp:2595
[17] wxGUIEventLoop::PreProcessMessage       	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\evtloop.cpp:145
[18] wxGUIEventLoop::ProcessMessage          	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\evtloop.cpp:163
[19] wxGUIEventLoop::Dispatch                	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\evtloop.cpp:230
[20] wxEventLoopManual::ProcessEvents        	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\evtloopcmn.cpp:228
[21] wxEventLoopManual::DoRun                	c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\evtloopcmn.cpp:268
</details>

closed 2019-06-24 by YuSanka

YuSanka, 2019-06-24:
Thanks for your report. This bug was fixed and will be a part of an upcoming release.

mloidl, 2019-06-25:
@YuSanka: 
I can confirm the crash is fixed. 
But i want to mention that the behavior is slightly different as with the official build.
When right clicking while dragging the upper slider the bottom slider moves up to the cursor in the official build. This does not happen when i build the current master-branch by myself.

YuSanka, 2019-06-25:
@mloidl, thanks for your testing and observations ;) 
Actually, this is now the expected behavior. Left click is used for separate slider's thumb moving. Right click switches you to the "one layer" mode and both thumbs move at the same time.

mloidl, 2019-06-25:
@YuSanka: I'll test it again when i'm at home. But what i can remember the switch to "one layer" mode did not work.

mloidl, 2019-06-26:
@YuSanka: Everything seems to be. Seems is previously did not understand how the "one layer" should exactly work. But after some playing it makes totally sense how it's implemented.

commit 14d8621f (2019-06-24): Color_print issues :
 - fixed #1933
 - implemented thumb moving to the mouse click position
 - implemented "discard color changes" button