prusa3d/PrusaSlicer #1936: Supports don’t print

andresdrb, 2019-03-12:
### Windows 10
_

### Geeetech A20M on 1.1.8 Marlon

### Behavior
* when printing supports or rafts, it seems to no extrude at all. The rest of the print extrudes perfectly, but when it gets to print supports it doesn’t extrude. 
* how to reproduce:  with supports from plate or raft
  
* expected result: supports correctly printed
* result: no supports
  * video about the problem:  https://1drv.ms/v/s!AmjNYeWWIj5pjXpHZdjbCk2oVOyc

#### Project File (.3MF) https://1drv.ms/u/s!AmjNYeWWIj5pjXm3NoHvwqqjd_ia

bubnikv, 2019-03-17:
I don't know how we can help you. I just checked with the G-code preview of Simplify3D and the support extrusions are there. Most likely either your firmware or the bowden setup does not like the way the supports are extruded by Slic3r PE. To me it looks like your extruder is clogging. This would not happen with the Original Prusa i3 MK3S :-)

bubnikv, 2019-04-13:
No feedback over a month, closing.

closed 2019-04-13 by bubnikv