prusa3d/PrusaSlicer #1937: GTK+3 support

Salamandar, 2019-03-12:
Hi,
As I was updating the PKGBUILD for Archlinux, I tried to use Gtk+3… And it works fine !
I'm tracking master, so a lot is missing (due to the c++ port), but no bugs found yet. Apart from a lot of GTK warnings, of course (mostly negative content width/height errors).

Here is the patch used : 
```
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index b38ae3665..d7d40c65e 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -60,6 +60,7 @@ if (SLIC3R_GUI)
     endif()
 
     if (CMAKE_SYSTEM_NAME STREQUAL "Linux")
+        set (wxWidgets_CONFIG_OPTIONS --toolkit=gtk3)
         if (SLIC3R_WX_STABLE)
             find_package(wxWidgets 3.0 REQUIRED COMPONENTS base core adv html gl)
         else ()
```

bubnikv, 2019-03-17:
I suppose we should set up our build system to compile against both the GTK2 and GTK3, and to provide both binaries to the public. I know the GTK2 themes are messed up on the Ubuntu 18 (or at least they were), so compiling against GTK3 will likely improve the situation on Ubuntu 18.

Salamandar, 2019-03-28:
Nice !

vojtechkral, 2019-03-28:
Glad you like it, but I should also add the option in the `deps` scripts and document it in `doc` (reopening to track that)...

commit ba66f34c (2019-03-25): Build: Add an option to choose GTK version on Linux, fix #1937