prusa3d/PrusaSlicer #1938: [Feature Request] Aligned rectilinear infill

blackyarr, 2019-03-12:
**Version**
1.41.3+

**Reason for request**
Ability to disable crosing infil lines. 

Useful when you print transparent objects or special test objects.

**feature is available in slic3r**
https://github.com/slic3r/Slic3r/issues/3225
https://github.com/slic3r/Slic3r/pull/3518

CCMCAGP, 2019-08-20:
I second this. This feature is very useful for mechanical testing, and at the moment, it's just simpler to use Cura which does this really easily. I don't want to revert to Cura, but I don't know how to implement the fork on my version of Slic3r which is 2k+ commits ahead,  and besides I'm using PrusaSlic3r. It would be really cool if one day this becomes implemented. Definitely useful for a scientific approach (dogbone tests). 
So +1, hoping this gets sorted by a benevolent soul soon!

wavexx, 2020-09-13:
I'd love this. I had a couple of scenario that needed finer control over the alternate angle.
I'll comment here instead of inside #2856.

It works, but adding *another* infill type is less than optimal considering how many we start to have.
It would be nice to have a new setting to control the internal behavior of FillRectilinear. A simple approach: a new "alternate angle", which set at 90 deg to match the current behavior.

What I wished this setting would *also* do is to counter-rotate the fill pattern by "alternate angle" / 2. "Pattern divergence" might be a better name for this.

The reasoning: let's say I want to align the fill pattern along X, spread by 10C: set fill angle to 0, divergence to 10.
First layer is rotated by -5, second layer  by +5. Makes more sense than setting "fill angle to -5" and "alternate angle to 10".
Such a pattern will yield ensure better adhesion between layers while still favoring strength along a main axis.

neophyl, 2020-09-13:
Hey Wavexx, it may sound like I’m harping on but in the last release of Super Slicer Merrill added just such a function, you can define the initial angle and now also a rotation angle for subsequent layers. Several of us asked for it.

wavexx, 2020-09-13:
I love SuperSlicer, but I secretly wished Merill would attempt to do PRs so that we could get the best of both :/
I know this takes a ton of time, so I guess that's the main reason we're stuff with two forks..

bubnikv, 2020-11-21:
Implemented with f0606c2a76da890dea359a9111825bbdb736a799 based on 
https://github.com/prusa3d/PrusaSlicer/pull/2856

closed 2020-11-21 by bubnikv

5axes, 2021-01-08:
tested wtih PS 2.3.0 Rc2 & Rc3 no difference between  **_Aligned rectilinear_**  anb **_Rectilinear_** ?