prusa3d/PrusaSlicer #1939: Code signing certificate has been revoked error in Windows 10

njcrawford, 2019-03-12:
### Version
1.41.3+win64

### Operating system type + version
Windows 10 Home

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* Download drivers from http://prusa3dprinters.vshcdn.net/downloads/drivers/prusa3d_win_2_2_3.exe
* Run prusa3d_win_2_2_3.exe
* Expected installer to open and show installation wizard
* Windows shows a file blocked message instead [image]

### Additional information
This isn't happening on all PCs, at least not yet. I have 2 that are showing this error and 2 that aren't. There's another user on Facebook having the same problem as well: https://www.facebook.com/groups/prusacommunity/?multi_permalinks=957685964572136&notif_id=1552186925226374&notif_t=group_highlights

A workaround suggested on the Facebook post is to start CMD as administrator and run setup from there. This worked for me on 1 PC, but I don't have access to the other PC to test it right now.

This is what Windows says if I dig down into the file properties to view the certificate details:
[image]

njcrawford, 2019-03-12:
I've found that you can force this issue to occur on a PC by clearing the certificate revocation cache. If you open a command line window and run `certutil -urlcache * delete` the PC will start showing the file blocked message.

bubnikv, 2019-03-12:
I checked with the college, who creates our windows installer. They are aware of the issue, they are in the process of buying a new certificate from a bullet proof certification authority.

N3X15, 2019-03-12:
Workaround:

1. Install a Windows 10 SDK from Microsoft.
1. Find `signtool.exe` and copy its path to clipboard. I recommend using [Everything](https://www.voidtools.com/) for this.
1. Run: `"path\to\signtool.exe" remove /c prusa3d_win_2_2_3.exe`

This will strip the codesigning certificate from the EXE.

bubnikv, 2019-03-14:
I am closing this issue, it is not connected to the Slic3r PE project, but to the Prusa3D drivers installer, and they are working diligently to fix it.

closed 2019-03-14 by bubnikv