prusa3d/PrusaSlicer #1940: [Request] Multi Material set extruder number on multiple items at the same time

zehkaiser, 2019-03-12:
When selecting the extruder number in the object list it would be nice to set the extruder number on multiple items at the same time. The list box already supports multi-select, but clicking the extruder to enable the drop down box forces the selection to a single object. Maybe a context menu could allow setting the extruder without needing to use the drop down box?

YuSanka, 2019-03-13:
Thanks for your request :) Working on it in the [SPE-861](http://dev.prusa3d.com/browse/SPE-861)

bubnikv, 2019-03-17:
It now works with clicking on the "wheel" icon with the right mouse button to open the pop up menu. Now you will find a new "set extruder for selected items" command there.

closed 2019-03-17 by bubnikv