prusa3d/PrusaSlicer #1941: Slicing Error - parts of model missing

Wot-Dabny, 2019-03-12:
### Version
_Version of Slic3r Prusa Edition used goes here_
1.41.2+win64

### Operating system type + version
Slic3r Prusa Edition
Version: 1.41.2+win64
Build:   Slic3rPE-1.41.2+win64-full-201811221420
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 10.0.17763.316]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    D:\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0\MSWin32-x64-multi-thread
    D:\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0
    D:\Prusa3D\Slic3rPE\local-lib\lib\perl5\MSWin32-x64-multi-thread
    D:\Prusa3D\Slic3rPE\local-lib\lib\perl5
    D:/Prusa3D/Slic3rPE/lib
    .
OpenGL installation
GL version:   4.6.0 NVIDIA 418.91
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 980M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
[... 361 lines cut ...]
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
i3 mk3 latest firmware, not sure of number

### Behavior
* _Describe the problem_
When I slice a downloaded .stl pieces disappear - the model is solid in the 3d view, but in the preview after slicing there are sections missing - see attached .jpgs
* _Steps needed to reproduce the problem_
Import .stl, slice
* _Expected Results_
whole object rendered
* _Actual Results_
pieces of object missing
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]
[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_
Can't upload - get error "We don't support that file type. Try again with a GIF, JPEG, JPG, PNG, DOCX, GZ, LOG, PDF, PPTX, TXT, XLSX, or ZIP." - so I zipped then attached it
[R_Ion_Accelerator_2b_fixed_0.1mm_PLA_MK3.zip](https://github.com/prusa3d/Slic3r/files/2957956/R_Ion_Accelerator_2b_fixed_0.1mm_PLA_MK3.zip)

bubnikv, 2019-03-13:
thanks for the report. Fixing with netfabb solves your issues (fixing through netfabb is integrated on Windows 10)

[R_Ion_Accelerator_2b_fixed-by-netfabb.zip](https://github.com/prusa3d/Slic3r/files/2960315/R_Ion_Accelerator_2b_fixed-by-netfabb.zip)

This is a screenshot of the 3D tool free, showing open edges (cracks)
[image]

This is a screenshot from meshlab (looking at the bottom), showing flipped faces with dark color
[image]

Your object contains multiple internal surfaces and overlapping bodies, here I used the 3d-tool to get a cross-sectional view
https://www.3d-tool.com/en_cad-viewer-download.htm

[image]

Such issues are very tricky for Slic3r to handle, though the Netfabb fixing service is smart enough to remove these internal walls. We should try to improve, but for now we recommend you to use the Netfabb for such models.

By the way, where did you get the model from? The model seems to be targeted for computer games / computer animation, not for 3D printing.

Wot-Dabny, 2019-03-14:
Thank you for the info and explanation. I got the model off of Thingverse - it is a model of a Warhammer Mech - Tabletop role playing game.Sean
Somewhere between physics and philosophy is the intersection of the real world. 

 
 

https://www.3d-tool.com/en_cad-viewer-download.htm

closed 2026-09-29 by prusaslicer-bot[bot]