prusa3d/PrusaSlicer #1943: Input fields are not working correctly

3eggert, 2019-03-13:
### Version
I'm using
1.42.0.alpha5+linux64
now but this issue is there since a long time. Meanwhile I got a new machine and reinstalled the OS several times, and the version of the OS also changed of cause. 

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Xubuntu 18.10

### Behavior
*  I can't delete data in input fields, I can't use cursor keys or return in input fields, mouse pointer disappears over input fields 
* open for example printer settings->custom gcode click in start g-code try cursor keys, del or backspace 
* possibility to enter new lines, delete characters 
* noting happens. No new lines are added or characters are deleted

I do not have such issues with any other programs on my machine

bubnikv, 2019-03-14:
please retest with alpha7
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha7

3eggert, 2019-03-14:
I tried, it makes no difference, as mentioned I have the issue quite a long time, since more than a year. 

I just recognized  some strange behavior, the not working keystrokes are made in the moment when I click on an other window for example a xterm (change focus away from slic3r window)

bubnikv, 2019-03-17:
This is interesting, as you are the only one reporting this issue.

Could you make a video, please? I wonder whether somebody will chime in with a similar problem, or whether you are the only one experiencing the issue and your Linux box configuration is somehow special.

3eggert, 2019-03-17:
Yes I agree, its very strange, I work more than 10 years as developer mostly exclusive on Linux an and never experiences such behavior.
It seems videos can't be uploaded, so here is a link to the video:

https://mega.nz/#!5gYwnATa!0JQNPLZqikTyPzybKRuXNkjBK0MpQ1DzybY58ZgDsj4

bubnikv, 2019-03-17:
Gruesse nach Bayern (ich glaube das Akcent richtig ausgewertet).

This is really weierd.

When you are a programmer, are you able to compile against wxWidgets 3.1.1
/ GTK3? I know Ubuntu has the GTK2 messed up in some way.

Do you have any plugin or whatever there may be to customize your keyboard
handling? As you are German, at least you likely have the German keyboard
installed.

3eggert, 2019-03-18:
Good morning
near miss, right country but not deep in the south, far in the north. I'm from Hamburg (-:

I tried the latest 1.42.0 beta as well as I compiled two own versions. One with the system wxWidgets and one with wx3.1.1 (which I double checked with ldd). Sadly it changed nothing. 

Yes, you are right I use german eliminate dead keys, and as I'm writing this I check for tools like you mentioned. I found out that I 'use' this tool:
https://github.com/ibus/ibus  
in version 1.5.19
'use' means it got installed but I never actually used it or was aware of it. 
And guess what, after killing that thing the input fields do as one would expect

Thanks a lot for helping me, I just checked the oldest version Slic3r lying around on my disk is from Okt 21 probably I was fighting with this even longer.

And now I checked KiCad, an other wx based application, an there also the ibus thingy breaks the input fields in the same way.

Once again many thanks and have a nice week.

Greetings

Thorsten

closed 2019-03-18 by 3eggert

3eggert, 2019-03-18:
As I now know what to search I found out I'm not alone. Googleing "ibus wxwidgets" gives 14600 results, for example a bug report for KiCad from 2014-10-24 which has a comment with a fix for it (https://bugs.launchpad.net/kicad/+bug/1385020/comments/13)

### The fix is:
```
sudo apt install ibus-gtk
sudo apt install ibus-gtk3
sudo reboot

```
Now it works for me

bubnikv, 2019-03-18:
great, thanks for the report.

bubnikv, 2019-03-18:
I have added your notes to the wiki
https://github.com/prusa3d/Slic3r/wiki/Known-Issues-and-Solutions
Please feel free to extend the section with more details if you feel to.