prusa3d/PrusaSlicer #1944: Wipe tower is not generated

wilecoyote2015, 2019-03-13:
### Version
1.42.0 Alpha 7

### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Zonestar P802QR2, Marlin firmware

### Behavior
Although the wipe tower is enabled in the print settings, wipe towers are neither shown nor printed.  

Steps to reproduce:
1. Add an object to the plater
2. Assign Extruder 2 to the support material and Extruder 1 to the other feature types in the print settings
3. Set the Contact Z distance to 0 in the print settings
4. Enable the wipe tower in the print setting
5. Slice the object

#### Project File (.3MF) where problem occurs
[untitled-Cube.zip](https://github.com/prusa3d/Slic3r/files/2963008/untitled-Cube.zip)

bubnikv, 2019-03-17:
Wipe tower is currently only supported for Single extruder multi material due to the limitations of the development resources.

tcm0116, 2019-05-01:
@bubnikv - I've started looking the code to see how the wipe tower could be generalized for use with all multi-extruder configurations. I'm thinking that I'll separate the MMU-specific tool changing operations from the tower generation. This would allow the tower to be generated using standard tool changes for non-MMU configurations or using MMU-specific tool changes for MMU configurations.

If you have any thoughts on how you'd like to see this accomplished, it would be great if you could share them so that I can work towards something that you would be willing to incorporate.

lukasmatena, 2019-05-01:
@tcm0116 Thanks for looking into the topic. It was me who attempted to make the wipe tower more generic a year ago. And although I would have to remind myself first, I will write my 2 cents. @bubnikv will certainly add what I forgot or didn't realize.

The most important thing to do is to decide and summarize what should and what should not actually happen with multiple extruder printer. This is a task for someone familiar with multiextruder printer (which I am not) and ideally with our MMU as well to cover all the differences. I was hoping to start a discussion about this within https://github.com/prusa3d/Slic3r/issues/718, but never had time to dive into it.

Anyway, I think the wipe tower is now generic enough to cover most of the differences by parameters that are already there. E.g., ramming can be turned off, loading distance and parking position set to zero and cooling moves can be eliminated, all that without ever touching the code. The settings are under Printer Settings->Single extruder MM setup and Filament Settings->Advanced->Toolchange parameters with single extruder MM setup (i.e. marked as SE specific).

So the strategy could be such: The wipe tower constructor loads all the parameters. In case it is SE printer, it would do what it currently does. Otherwise, it will overwrite its copies of the parameters with whatever is needed to turn off the unwanted stuff. I would not touch the rest of the code unless needed, it is quite brittle. It would be best to refactor it, but I decided not to fix what's not broken and I wouldn't do it now.

At least some code changes will be needed though. For instance, one must make sure that the Custom toolchange GCode is injected where appropriate (it is currently ignored if wipetower does the toolchange - https://github.com/prusa3d/Slic3r/issues/1245 - and maybe even that should behave differently). Also, there is a debate about when the temperatures should be switched (https://github.com/prusa3d/Slic3r/issues/1339, https://github.com/prusa3d/Slic3r/issues/1564), I think our testers tried to devise something better than we currently have, but without conclusive results. Similar debate may arise with multiple extruder printers and the code will have to be altered to cover it.

I would be very careful with modifying the code, I remember it's more spaghetti than it looks. Also, PRs with large diffs are more difficult to evaluate. If you decide to do it, I suggest changing as little as possible and [tread softly](https://www.wolfson.ox.ac.uk/~ben/writings/TheClothsofHeaven.html).

Thanks again for volunteering.

tcm0116, 2019-05-01:
> Anyway, I think the wipe tower is now generic enough to cover most of the differences by parameters that are already there. E.g., ramming can be turned off, loading distance and parking position set to zero and cooling moves can be eliminated, all that without ever touching the code.

That was my initial assessment of the code. However, I noticed that a fixed 15 mm is added to the total retraction distance during the retraction phase of the unloading sequence.  I think this value should be made configurable as some extruders may not tolerate the 15 mm retract and restore process that happens if all other values are zeroed out.

https://github.com/prusa3d/Slic3r/blob/4a5992ba6eca8e1ddb1e576a009660c80b910d1a/src/libslic3r/GCode/WipeTowerPrusaMM.cpp#L844-L849

> So the strategy could be such: The wipe tower constructor loads all the parameters. In case it is SE printer, it would do what it currently does. Otherwise, it will overwrite its copies of the parameters with whatever is needed to turn off the unwanted stuff.

I've already modified the code to enable generation of the tower if the "Single Extruder Multi Material" option is disabled. I am having an issue with the tower not rendering after enabling the wipe tower in the settings, but it does render if a part is loaded (or reloaded from disk). I need to see if this occurs in the unmodified Slic3r.

> At least some code changes will be needed though. For instance, one must make sure that the Custom toolchange GCode is injected where appropriate

The current change I have is that if "Single Extruder Multi Material" is enabled, then it will not use the cusutom toolchange gcode, but it will use it if "Single Extruder Multi Material" is disabled.

> PRs with large diffs are more difficult to evaluate. If you decide to do it, I suggest changing as little as possible and tread softly.

What would you think about me going ahead and creating a PR that would include the changes as I make them? This would give folks a chance to make any comments or suggestions as I make progress.

lukasmatena, 2019-05-01:
>  I noticed that a fixed 15 mm is added to the total retraction distance

It is not added, it's to make sure that the initial unload is done at high speed and independent on the unloading distance. The total retraction distance is not affected. But you are right, this is another thing that probably needs to be changed. I would personally keep it as it is for SE and just skipped it for ME if that makes any sense. There may be other things like that.

> The current change I have is that if "Single Extruder Multi Material" is enabled, then it will not use the custom toolchange gcode, but it will use it if "Single Extruder Multi Material" is disabled.

I believe that's what our current master does.

>  I am having an issue with the tower not rendering

Conditions like (more_than_one_extruder && se_printer) are scattered on several places. It's part of the job to go through them and think about them one by one. They are on more critical places then just rendering.

> What would you think about me going ahead and creating a PR

I would give you a green light and wished you good luck. But I'm not the project lead and the workload of processing PRs and taking responsibility for them is not on my shoulders.

tcm0116, 2019-05-02:
> It is not added, it's to make sure that the initial unload is done at high speed and independent on the unloading distance. The total retraction distance is not affected.

For the gcode sample below, I set all of the printer MMU and filament toolchange parameters to 0 (including the ramming settings). As can be seen in the toolchange unload section, the filament is retracted 15 mm and then restored incrementally, exactly as the code segment I referred to above reads. The toolchange load section also has unneeded commands that can be removed.

```
;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : PLA
;--------------------
M220 B
M220 S100
; CP TOOLCHANGE UNLOAD
G1 X19.250 Y44.829 
G1 E-15.0000
G1 E10.5000
G1 E3.0000
G1 E1.5000
M104 S190
G1 F2000
G1 Y44.279 F2400
G4 S0
T1
G4 S0
; CP TOOLCHANGE LOAD
G1 F3000
G1
G1
G1 Y44.779
; CP TOOLCHANGE WIPE
G1 X-19.750 E2.1350 F800
G1 X-20.000
G1 Y45.379
G1 Y26.279 E0.8861
G1 X20.000 E1.8557
G1 Y45.379 E0.8861
G1 X-20.000 E1.8557
M220 R
G1 F6000
G4 S0
G92 E0
; CP TOOLCHANGE END
;------------------
```

tcm0116, 2019-05-03:
I've submitted PR #2198 to enable the wipe tower for all multi-extruder configurations.

rtyr, 2020-04-17:
Implemented. Closing.

closed 2020-04-17 by rtyr